chuot/rdio-scanner #124: Enhancement: Android Auto and CarPlay Support

jslitzkerttcu, 2022-02-09:
Please consider adding android auto and car play support to the native apps.  This was added to https://scannerradio.app  for Broadcastify feeds and would be very valuable to have similar functionality for the native apps with RdioScanner.

blackwellj, 2022-02-28:
I second this for CarPlay! It would be so useful!

atlex, 2022-03-01:
Android Auto vote :)

chuot, 2022-03-07:
I am working on a new audio subsystem for the upcoming native v1.5 where this enhancement request will be addressed.

jslitzkerttcu, 2022-03-08:
> I am working on a new audio subsystem for the upcoming native v1.5 where this enhancement request will be addressed.

Nice!  Looking forward to it!  If you need beta testers, I have iOS and Android and am happy to test for you!

adorsett, 2022-03-14:
I third this request for CarPlay support.