chuot/rdio-scanner #126: IP Bind

JayManCanada6014, 2022-02-10:
Hello,

I use TailScale to let my VPS's talk including a reverse proxy.   On v5 when I would start Rdio-Scanner it would bind like this.
0.0.0.0:3000            0.0.0.0:* 

on v6
*:3000                 *:* 

v6 my reverse proxy cannot connect, but on v5 it does and these are the only two differences.  Can we set the bind ?  I tried to do -listen 0.0.0.0:3000 but it still only took the port and not the IP

Dewey3, 2022-02-10:
Will `"-listen string"` on page 7 of the PDF help?

JayManCanada6014, 2022-02-10:
I tried  -listen 0.0.0.0:3000 which would be the formatting but it didn't work.   Only -listen :3000 seems to do anything that that puts in *:3000

chuot, 2022-02-10:
-listen :3000 (which is the same as not specifying it because its the default) will bind to all the available network interfaces on the host.

I don't know how TailScale works, but I guess it creates a virtual network interface, which must exists when Rdio Scanner starts.

Also, I'm not sure how to interpret your `0.0.0.0:3000 0.0.0.0:*`. Should it be `*:3000 *`? Again, I don't work with TailScale and don't know much about it (yet).

JayManCanada6014, 2022-02-10:
I closed it, I figured out the solution