chuot/rdio-scanner #129: Crashing/frozen 6.1.10 after 10-15mins on Ubuntu 20.04.3 LTS

canuckcam, 2022-02-13:
I upgraded to 6.1.10 on Ubuntu 20.04.3 LTS a few days ago, and started to have this problem yesterday where after 10-15mins, rdio-scanner will not show the web interface or playback in the app. I can't seem to restart the service either (-service restart or stop), it doesn't have any errors and puts me back into to the $ prompt. Once I do that, and look at netstat | grep rdio-scanner, I still see port 80 taken. If i run ./rdio-scanner interactive, it errors about unable to bind to port 80.

This is an instance of only API feeds, no DirWatch. I'm neither out of memory (31% usage) or disk space (20GB free) or CPU (0.73 load). The .db file is 2GB in size, so I thought that might be an issue, so I deleted it, restored config from a backup, but it happened again, db file size is approx 730MB.

in journalctl I found this:

```
Feb 12 19:53:28 rdio-scanner rdio-scanner[710]: 2022/02/12 19:53:28 http: Accept error: accept tcp [::]:80: accept4: too many open files; retrying in 10ms
Feb 12 19:53:28 rdio-scanner rdio-scanner[710]: 2022/02/12 19:53:28 controller.ingestcall: call.write: unable to open database file (14)
Feb 12 19:53:28 rdio-scanner rdio-scanner[710]: 2022/02/12 19:53:28 http: Accept error: accept tcp [::]:80: accept4: too many open files; retrying in 5ms

```

Eventually I do see (but I'm not out of memory?)

`Feb 12 19:53:38 rdio-scanner rdio-scanner[710]: 2022/02/12 19:53:38 controller.ingestcall: call.write: unable to open database file: out of memory (14)
`

canuckcam, 2022-02-13:
Update: happens as well on 6.1.11. Empirically, it seems all is well as long as someone is connected/listening. I left it on in the background for an hour and it was fine. But as soon as I closed the browser tab and reloaded a few minutes later, gone.

JayManCanada6014, 2022-02-14:
I'm testing 6.1.12 right now I will let you know how it goes, I was having the same above issue on 6.1.11

JayManCanada6014, 2022-02-14:
6.1.12 Almost an hour now without it freezing.

JayManCanada6014, 2022-02-14:
7-9:30pm on 6.1.12 without issues.  And still going.

JayManCanada6014, 2022-02-14:
Ran all night and at 12hours still no issues.

chuot, 2022-02-14:
v6.1.12 now released with a fix to this issue.

@JayManCanada6014 remember to install this new release over your test release.

canuckcam, 2022-02-15:
Unfortunately it didn't solve the issue for me. :( I'm not sure what changed between the versions, but I slowly went back each version to .7 and it still happens. My install is personal, so there's only a few people using and it may have failed earlier but just didn't know although with .12 it seems to have taken an hour or longer to stop responding.

JayManCanada6014, 2022-02-15:
> Unfortunately it didn't solve the issue for me. :( I'm not sure what changed between the versions, but I slowly went back each version to .7 and it still happens. My install is personal, so there's [...]

Yeah I am testing a new version for @chuot   already.        I also started using --max-time 30 in my API curl script so when it fails I don't get massive about of comms backing up on TR.  If it can't get to Rdio it scraps it after 30 seconds.

canuckcam, 2022-02-15:
A watched pot never boils. After just shy of 2 hours, it finally.
$ sudo journalctl -u rdio-scanner.service -f
[image]

chuot, 2022-02-15:
Next time this happens, could you do a `lsof -c rdio-scanner` followed by a `netstat -np --ip` and post the results?

JayManCanada6014, 2022-02-15:
@chuot    Just died after your email, I replied back with logs and what I see it doing.
Not going to post the logs here but this is what I seen.

Front End UI still answers,  nothing displaying NO LINK, but no calls play.
API upload is not working, connections are timing out after 30 seconds (my curl)
CAN Playback calls already in the system

stop and restart and away we go again.

Load related?  I have 30 people on but it is a busy time of the day for radio traffic.

canuckcam, 2022-02-22:
Yes 6.1.13 has been stable for over a day. Thank you!

closed 2022-02-22 by canuckcam