chuot/rdio-scanner #131: http: TLS handshake error  tls: client offered only unsupported versions: []

photounit, 2022-02-16:
I have just upgraded to 6.1.12  and I had been having persistent lockups with the too many files oper error. I was not monitoring any folders and so far since upgrading to V 6.1.12, I have not had that error reoccur.  Now I am having an issue with this error - 

"2022/02/16 09:06:25 http: TLS handshake error from 195.230.103.241:51472: tls: unsupported SSLv2 handshake received
2022/02/16 09:06:26 http: TLS handshake error from 195.230.103.241:52190: tls: client offered only unsupported versions: []

Any ideas?

chuot, 2022-02-16:
Golang does not support SSLv2 negotiation, the client (old browser maybe) tries to negotiate SSLv2, which is now considered insecure as of now.

Unfortunately, I don't have control over these http:TLS handshake errors, but you can ignore them.

photounit, 2022-03-04:
Thanks for your help on this issue Chrystian....

closed 2022-03-04 by photounit