chuot/rdio-scanner #132: Reverse proxy does not work after upgrading from v4 to v6

buhalo1, 2022-02-19:
Hello. Until recently, I used rdio-scanner v4, running on a dedicated host (192.168.1.222),
access to the host from the internet was provided through a web server running on a separate host (192.168.1.111),
using a reverse proxy.
After upgrading to version 6, the reverse proxy stopped working properly. The frontend works,
but a "no link" warning is displayed at the top of the UI screen,
it is obvious that the problem is in some changes to the web socket.

Settings from config files:

config.json (v4):

"nodejs":
{ "env": "production",
"host": "192.168.1.222",
"port": "3000",
"sslCA": null,
"sslCert": null,
"sslKey": null }

rdio-scanner.ini (v6):

db_file = rdio-scanner.db
db_type = sqlite
listen = 192.168.1.222:3000

Apache directives:

RewriteEngine On

RewriteCond %{HTTP:CONNECTION} ^upgrade$ [NC,OR]
RewriteCond %{HTTP:UPGRADE} ^websocket$ [NC]
RewriteRule ^/?(.*)$ ws://192.168.1.222:3000/$1 [P,L]

RewriteRule ^/?(.*)$ http://192.168.1.222:3000/$1 [P,L]

Could you tell me what needs to be corrected in the above settings?

Thanks for the great app.

chuot, 2022-02-21:
Have you tried with the latest v6.1.13? The reverse-proxy issue has been fixed since v6.1.11.

buhalo1, 2022-02-23:
I am very puzzled, but after installing v6.1.13 a few hours after the release, I found that the reverse proxy still does not work. The settings are no different from those in which v4 worked without issues.

chuot, 2022-02-23:
Here's my config, which looks a lot like yours:

```
RewriteEngine On

RewriteCond %{HTTPS} off 
RewriteCond %{REQUEST_URI} !^/\.well\-known/acme\-challenge/
RewriteRule ^(.*)$ https://%{HTTP_HOST}%{REQUEST_URI} [L,R=301]

RewriteCond %{HTTP:CONNECTION} ^upgrade$ [NC,OR]
RewriteCond %{HTTP:UPGRADE} ^websocket$ [NC]
RewriteRule ^/?(.*)$ ws://blackbox:8006/$1 [P,L]

RewriteCond %{REQUEST_FILENAME} !-d
RewriteCond %{REQUEST_FILENAME} !-f
RewriteRule ^/?(.*)$ http://blackbox:8006/$1 [P,L] 
```
It is working like a charm with 6.1.13, so there must be something somewhere with your setup.

Is v6 running on the same host as the old v4?
Which version the webapp is showing when accessed from the through the proxy, v6.1.13?
Is http://192.168.1.222:3000/ working when accessed from you lan?

Also, some logs from Apache or from the javascript console (F12) should give us some hints. Please post relevent logs if any.

chuot, 2022-03-15:
It as been confirmed to work. Closing

closed 2022-03-15 by chuot