chuot/rdio-scanner #136: Downstream talkgroup select keeps reverting to all talkgroups 6.1.14

kb2ear, 2022-02-27:
I'm trying to select a subset of talkgroups from a system to upload. I go to the admin screen then downstreams then select the downstream I want to modify. I then click Choose Systems. then uncheck everything and select only groups I want to go downstream click on and the box says "This downstream allows access to some systems and talkgroups". I scroll down to the save button and hit it.

After it saves I go back to the downstream system and it says "This downstream allows access to some systems and talkgroups" and everything is checked again.

6.1.14 with mariadb backend.

chuot, 2022-02-28:
Confirmed.

chuot, 2022-03-07:
Should be fixed in 6.1.15

kb2ear, 2022-03-12:
It does indeed appear to be fixed in 6.1.15, thanks Chrystian.

closed 2022-03-15 by chuot