chuot/rdio-scanner #137: IOS Native 1.3.2 - Playing Avoided Channels

tadscottsmith, 2022-02-28:
The IOS native app seems to be playing channels that have been avoided. The AVOID label shows on the screen, but the audio is still played. Pressing the AVOID button twice to disable and re-enable AVOID will skip the currently playing call, but will not avoid future calls.
[image]

chuot, 2022-02-28:
There have been some fixes in version 1.4.1 and 1.4.2, but there's stil some issue on which I'm working on at the moment. Stay tuned for the upcoming release. Thanks

chuot, 2022-03-03:
Version 1.4.7 is out with fixes in regards to avoid/patch talkgroups. If a talkgroup is avoided but is patched to another talkgroup that isn't avoided, the `patch` flag will be displayed and those talkgroups won't be cleared from the queue.

tadscottsmith, 2022-03-07:
Seems to be working great. Thanks!!

closed 2022-03-07 by tadscottsmith