chuot/rdio-scanner #141: DB Migration from 5 to 6 Fails: UNIQUE Constraint Failed

adorsett, 2022-03-14:
```
2022/03/14 20:53:21 Preparing for database migration
2022/03/14 20:53:21 running database migration 20210830092027-v6.0-rename-index
2022/03/14 20:53:56 running database migration 20211202094819-v6.0.2-alter-table
2022/03/14 20:53:56 running database migration 20220101070000-v6.1.0
2022/03/14 20:54:49 constraint failed: UNIQUE constraint failed: rdioScannerUnits.systemId, rdioScannerUnits.id (2067) while doing insert into `rdioScannerUnits` (`id`, `label`, `order`, `systemId`) values (19344, '11A N DISP', 13, 1)
```

adorsett, 2022-03-14:
It appears that I have multiple unit IDs that have the same name because well, they do.  Looks like it's the dispatch consoles...

chuot, 2022-03-15:
It was a bug in v5 where you shouldn't have multiple [systemId, unitId] with the same values. Clean up your configuration in v5 before migrating to v6. Alternatively, you can start from scratch, export your configuration from v5, and reload the configuration into v6.

adorsett, 2022-03-16:
Thanks, I did some digging and found a single unit-id that was duplicated in the units db column.  I did a manual DB update and was able to remove the offender and the migration ran cleanly.  Appreciate the nudge in the right direction.

closed 2022-03-16 by adorsett