chuot/rdio-scanner #144: v6.1.15 Stopping

Dewey3, 2022-03-16:
Running Rdio-Scanner v6.1.15 on a 4GB Raspberry Pi 4 with two other RPis feeding it via Trunk-Recorder plugin.  The last two days, Rdio-Scanner has just stopped on its own without any notice.  I've found out in the evening when I go to look at the recordings and notice that I cannot get into it via the interface and must SSH in the RPi and reboot, at which point, everything works again.  I don't know if the timing is coincidental, but yesterday, the last recording prior to my manual reboot was at 11:44 am, and today the last recording prior to my manual reboot was at 11:41 am.  I've looked at Rdio-Scanner's logs, and there is no hint of anything other than just the recordings.  I'll keep an eye on it and see if it continues, and if so, if the time of day is still around the same time.

chuot, 2022-03-16:
Without logs, I won't be able to help you much with this.

Please try to locate some hints from the system logs using the command `journalctl -r`.

Dewey3, 2022-03-16:
Thank you very much Chrystian.  Looking at the journal logs with `journalctl -r`, nothing is jumping out at me, in fact, it looks like everything is "normal" to my limited understanding.  I am attaching the logs for both days for between 11:00 and just after noon.  The noon login is an every 3-hour cronjob running from my Raspberry Pi server that checks in on all of my feeder RPis to check their health... throttled status, temperature, disk space.

[20220314.txt](https://github.com/chuot/rdio-scanner/files/8262364/20220314.txt)
[20220315.txt](https://github.com/chuot/rdio-scanner/files/8262366/20220315.txt)
[2022-03__Feeder.log](https://github.com/chuot/rdio-scanner/files/8262367/2022-03__Feeder.log)

chuot, 2022-03-16:
How do you run your rdio-scanner instance? It doesn't seem to be started by systemd. I think it would be better to use the `-service` as explained in the PDF document. Then we should be able to see the system and rdio-scanner logs with `journalctl`.

Unfortunately, I couldn't find anything in your files that could provide any clues.

Dewey3, 2022-03-16:
I've never run it as a system, but instead always run it from a `@reboot` cronjob, even in the older `node` days.  I tend to lean towards being an old dog that doesn't like new tricks and since that has always worked for me, it's always what I've done.  It sounds like running as a service is now the better thing to do... no problem... time to learn a new trick.

Dewey3, 2022-03-17:
Thanks again Chrystian.  I am now running Rdio-Scanner as a service and I'll see how that works out for me.  I had to use the older `service` command to install and run it, because I could get `systemctl` to work for Rdio-Scanner... am I doing something wrong?

chuot, 2022-03-21:
I assume you already had a systemd service unit for an older v5 of Rdio Scanner. Simply delete all `rdio-scanner.*` files under `/etc/systemd/system`, then `sudo sytemctl daemon-reload` to refresh the systemd configuration. After that, try again the instructions on page 6 of the PDF document in regards to installing a service instance.

closed 2022-03-30 by Dewey3