chuot/rdio-scanner #145: v6.1.5 @ Ubuntu 20.04 = High CPU

JayManCanada6014, 2022-03-16:
### Discussed in https://github.com/chuot/rdio-scanner/discussions/143

<div type='discussions-op-text'>

<sup>Originally posted by **JayManCanada6014** March 15, 2022</sup>
Good Evening All,

About a week ago started noticing every 24hours  the CPU starts going up by large chunks, each 24hours goes up more until I restart Rdio-Scanner.  I  think it was a Ubuntu patch that came out but unsure, Is anyone else seeing this?

[image]
[image]
</div>

chuot, 2022-03-16:
Ok I see, that's the HTOP way of showing things.

Anyway, I'm sure it's related to dead client connections not being released from the thread loop. I've changed the code a bit in this regard and I'm currently stress testing it on my dev box which is feed by my prod instance. It is serving between 1 and 2000 clients right now and no CPU spikes. I'll let the stress test run for a few hours and if nothing bad comes up I'll release v6.1.16 in the hope that it fixes your problem.

It's also interesting that you mentioned the issue appears around the same time as an Ubuntu patch. Unfortunately, all my production and development boxes use Fedora.

kb2ear, 2022-03-17:
Same issue on debian 10
running a bunch of instances here.

chuot, 2022-03-17:
After a full day of stress testing the upcoming release, I was unable to reproduce the symptom. Sure, the CPU was spiking, but that was when emitting incoming calls to 2000 clients, but then it back to normal. So I wonder if compiling the app with Go 1.18 (latest) with updated libraries did something. Anyway, I'll continue working on this today.

Next time this happens on your setup, please `kill -3` your instance and get all the logs it generates. If Rdio Scanner was launched by systemd, it should restart right away.

chuot, 2022-03-17:
I think I may have found the cause. Please test v6.1.16 if you're still having this issue.

JayManCanada6014, 2022-03-19:
Looks good, CPU is staying low as expected.   
[image]

JayManCanada6014, 2022-03-21:
Mar 20 21:16:11 rdio-scanner rdio-scanner[1410951]: 2022/03/20 21:16:11 listeners count is 15
Mar 20 21:16:21 rdio-scanner rdio-scanner[1410951]: 2022/03/20 21:16:21 listeners count is 15
Mar 20 21:16:42 rdio-scanner rdio-scanner[1410951]: 2022/03/20 21:16:42 newcall: system=100 talkgroup=8768 file=8768-1647825391_142080000-call_2.m4a success
Mar 20 21:16:44 rdio-scanner rdio-scanner[1410951]: 2022/03/20 21:16:44 newcall: system=2000 talkgroup=5 file=1647825400-1.m4a success
Mar 20 21:16:52 rdio-scanner rdio-scanner[1410951]: 2022/03/20 21:16:52 newcall: system=102 talkgroup=35680 file=35680-1647825402_141435000-call_364.m4a success
Mar 20 21:16:53 rdio-scanner rdio-scanner[1410951]: 2022/03/20 21:16:53 newcall: system=100 talkgroup=8768 file=8768-1647825398_141630000-call_3.m4a success
Mar 20 21:16:53 rdio-scanner rdio-scanner[1410951]: 2022/03/20 21:16:53 listeners count is 16
Mar 20 21:16:56 rdio-scanner rdio-scanner[1410951]: panic: concurrent write to websocket connection
Mar 20 21:16:56 rdio-scanner rdio-scanner[1410951]: goroutine 25 [running]:
Mar 20 21:16:56 rdio-scanner rdio-scanner[1410951]: github.com/gorilla/websocket.(*messageWriter).flushFrame(0xc00004ee98, 0x1, {0x110e940?, 0x0?, 0x0?})
Mar 20 21:16:56 rdio-scanner rdio-scanner[1410951]:         /home/chuot/go/pkg/mod/github.com/gorilla/websocket@v1.5.0/conn.go:617 +0x52b
Mar 20 21:16:56 rdio-scanner rdio-scanner[1410951]: github.com/gorilla/websocket.(*Conn).WriteMessage(0xc0000f4000, 0xc000123e60?, {0x110e940, 0x0, 0x0})
Mar 20 21:16:56 rdio-scanner rdio-scanner[1410951]:         /home/chuot/go/pkg/mod/github.com/gorilla/websocket@v1.5.0/conn.go:770 +0x139
Mar 20 21:16:56 rdio-scanner rdio-scanner[1410951]: main.(*Clients).Remove(0xc00004efb0?, 0xc0002a8000)
Mar 20 21:16:56 rdio-scanner rdio-scanner[1410951]:         /home/chuot/Dev/rdio-scanner/server/client.go:234 +0xd2
Mar 20 21:16:56 rdio-scanner rdio-scanner[1410951]: main.(*Controller).Start.func3()
Mar 20 21:16:56 rdio-scanner rdio-scanner[1410951]:         /home/chuot/Dev/rdio-scanner/server/controller.go:636 +0x11d
Mar 20 21:16:56 rdio-scanner rdio-scanner[1410951]: created by main.(*Controller).Start
Mar 20 21:16:56 rdio-scanner rdio-scanner[1410951]:         /home/chuot/Dev/rdio-scanner/server/controller.go:619 +0x30a
Mar 20 21:16:56 rdio-scanner systemd[1]: rdio-scanner.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
Mar 20 21:16:56 rdio-scanner systemd[1]: rdio-scanner.service: Failed with result 'exit-code'.

Just had it crash and here is the report.

chuot, 2022-03-21:
This issue is about high CPU usage, as the issue title suggests. Please open a new issue with your evidence.

Consider closing this issue if the high CPU usage is fixed for you. Thanks.

closed 2022-03-21 by JayManCanada6014