chuot/rdio-scanner #150: Unit name and number not pulled from SDRTrunk file

infecticide, 2022-03-25:
OS: Ubuntu 21.10
rdio-scanner version: 6.1.16
SDRTrunk version: 0.5.0-alpha6

The "artist" metadata tag in the MP3 file contains the unit ID + whatever alias name may have been given to it in SDRTrunk. 

EG: 40006 EMS - Dispatch

rdio-scanner doesn't seem to be parsing this and as a result, the unit (nor it's alias) never appears in the app.

I have created a Powershell script to copy/move the SDRTrunk recordings to another location in a more structured manner for my own organisational purposes.  Once I had rdio-scanner setup, I added some code to it in an attempt to make the file names more informative to rdio-scanner but it doesn't seem to be using the unit id from the file name either.

Mask: #DATE_#TIME_#SYS_#TG_#UNIT
File Name Eg: 2022-03-24_171400_City of Regina_153_1570.mp3

chuot, 2022-03-25:
dirWatch type=sdrtrunk don't ingest unit name from the MP3 TAGS as of now. I will try to fit a fix in the v6.2.0 that I'm supposed to release today. Hopefully I will have time to do so.

Note that the dirWatch mask is ignored if the dirWatch type is sdrtrunk. It works only for the default type. However, your mask must match the whole file name, which isn't the case in your example. The #SYS has to be the ID, not the name.

chuot, 2022-04-25:
Fixed in v6.2

closed 2022-04-25 by chuot