chuot/rdio-scanner #151: Docker Container keeps restarting after 6.2.0 upgrade

rml596, 2022-03-25:
After upgrading to `v6.2.0` my docker image keeps restarting, I can't figure out exactly what the issue is. 
It seemed like the image came up but then shortly after panicked and is now in this state.. 

I ran `docker-compose pull && docker-compose up -d`. I was on `v6.1.12` prior to the upgrade.

```
Rdio Scanner v6.2.0
----------------------------------
2022/03/25 13:27:04 server started
2022/03/25 13:27:04 base folder is /app/data
2022/03/25 13:27:04 main interface at http://151804b91795:3000
2022/03/25 13:27:04 admin interface at http://151804b91795:3000/admin
2022/03/25 13:27:06 listeners count is 1
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x18 pc=0x99c938]

goroutine 37 [running]:
main.(*Units).Merge(0x0, 0x9d14e0)
	/app/server/unit.go:97 +0x38
main.(*Controller).IngestCall(0xc0000d6370, 0xc0002879e0)
	/app/server/controller.go:320 +0x10bf
main.(*Controller).Start.func2()
	/app/server/controller.go:615 +0x46
created by main.(*Controller).Start
	/app/server/controller.go:612 +0x2d1

Rdio Scanner v6.2.0
----------------------------------
2022/03/25 13:27:22 server started
2022/03/25 13:27:23 base folder is /app/data
2022/03/25 13:27:23 main interface at http://151804b91795:3000
2022/03/25 13:27:23 admin interface at http://151804b91795:3000/admin
2022/03/25 13:27:24 listeners count is 1
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x18 pc=0x99c938]

goroutine 23 [running]:
main.(*Units).Merge(0x0, 0x9d14e0)
	/app/server/unit.go:97 +0x38
main.(*Controller).IngestCall(0xc00014e370, 0xc0002bf680)
	/app/server/controller.go:320 +0x10bf
main.(*Controller).Start.func2()
	/app/server/controller.go:615 +0x46
created by main.(*Controller).Start
	/app/server/controller.go:612 +0x2d1
```

docker-compose.yml
```
cat docker-compose.yml
version: "3"

services:
  server:
    container_name: rdio-scanner
    environment:
      - TZ=America/New_York
    #image: docker.io/chuot/rdio-scanner:latest
    image: 52b117e42700
    ports:
      - "3000:3000"
    restart: always
    volumes:
      - DIR/.rdio-scanner:/app/data:z
      - TR-Recordings-DIR:/recordings:z
      - TR-Recordings-DIR:/conv-recordings:z
```

Even though I have the recordings directory mounted, its not active as I am using the TR upload plugin.

Not sure if anyone else is seeing this.

chuot, 2022-03-25:
v6.2.1 should fix it.

chuot, 2022-03-25:
Not true, still missing half the fix. Wait for 6.2.2.

rml596, 2022-03-26:
It looks good. I will reopen if it pops back up. Thanks for the quick fix!

closed 2022-03-26 by rml596