chuot/rdio-scanner #158: Unable to Search Talkgroups

Dewey3, 2022-04-01:
I don't know if this is just me, but I've tried from three different computers at two different locations, using Firefox and Chrome.  When I try to search talkgroups, the talkgroup popup will appear for a very quick second, then collapse to "All Talkgroups" and not give anything else.   I noticed this when I went to v6.2.2 and seeing the same with 6.2.3.  Here is a very brief video of what I am seeing:  [https://drive.google.com/file/d/1ADhjRnKsAnkr2zBsff2epa6sUQuSjfgb/view?usp=sharing](url).

chuot, 2022-04-04:
Is there anything reported in the javascript console (F12) while you're having the issue?

Try to shift-reload (Shitft-F5) the webapp from the browser to see if the issue vanished by itself.

If possible, send me by email an exported config from the admin dashboard. It will test it on my side.

Thanks

chuot, 2022-04-04:
Upon further analysis, it appears that this issue is caused by the auto populate feature where all search filters are rebuilt when a new configuration is pushed to clients. I was able to reproduce this behavior. So instead of rebuilding search filters every time a new configuration is pushed to clients, I'll do it once, every time you open the search panel. I will include this change in the next version due out today. Thanks for reporting this issue.

chuot, 2022-04-04:
Give a try to v6.2.4, it should be fixed.

Dewey3, 2022-04-04:
Yup, you nailed it.  Thank you for everything AND for Rdio-Scanner!

closed 2022-04-04 by Dewey3