chuot/rdio-scanner #160: Unable to Make Configuration Changes

Dewey3, 2022-04-04:
The now fixed problem I reported in [https://github.com/chuot/rdio-scanner/issues/158](url) has now moved over to the admin panel.  When I attempt to make configuration changes in my Systems, the Systems drop-down boxes close as fast as they open preventing access to the System settings.  Since this appears to be exactly what was happening in [https://github.com/chuot/rdio-scanner/issues/158](url), I temporarily stopped my two Trunk-Recorder feeders from feeding Rdio-Scanner and the problem disappeared.  Once the feeders were restarted, so did the problem.

chuot, 2022-04-05:
Same as with discussion #154, it's by design. This is because auto populate is enabled and the admin dashboard refreshes its data when a new configuration is commited (via auto populate). The dashboard will not refresh its contents if there are pending changes, but if it is fresh it will refresh and open tabs will close, as it happens for you with the system tab. Either temporarily disable auto populate for the time you want to change the configuration and then re-enable it. This is not a problem on a system that has been auto populated for a long time, but I agree it can be annoying when auto populated intensively.

Dewey3, 2022-04-05:
Thanks Chrystian.  In my case, I believe this happens every time a new recording is fed to Rdio-Scanner (via Trunk-Recorder plugin from a different RPi).  Both of my systems are well established somewhere around or approaching two years.  The system with the dynamic 2##### talkgroups only receive "new" information once or twice a week, and even then, the dynamic talkgroup will remain the same for that day.  (I nip new dynamic talkgroups in the bud when they appear by adding them to the Trunk-Recorder csv file with a 255 priority).  So I believe what is happening in my case is that each time the very busy system sends a new transmission to Rdio-Scanner, it's triggering the reset, thus the resetting as soon as I open the drop-down.  Either way, I can work with it... momentarily turning off auto-populate is much easier than stopping both feeders... and will work than way forward.  Again, thank you sir!

chuot, 2022-04-05:
I just saw in the code that I'm not checking if the unit id/label (if any) already exists and I assume it should be auto-populated. I will change the code for the next version to check if the unit id already exists and if its label is the same, then I will not auto populate the same value. Therefore, the admin dashboard should refresh less often. you can reopen the issue, I will mark it as enhancement.

chuot, 2022-04-06:
Give a try to v6.2.5. It will still close admin panel, but only if unit id/label doesn't exist already.

Dewey3, 2022-04-06:
So far so good, everything is working perfectly.  Thank you again sir!

closed 2022-04-06 by Dewey3