chuot/rdio-scanner #162: Android App's Audio Stops But All Works Fine

noamlivne, 2022-04-11:
Sometimes the audio of the Android App stops for no apparent reason.
There is no issue with the Link, nor battery-conservation issue. The calls are continuing to play (but no sound) in the Live Stream, the Q is raising. All is fine besides no sound.
Swiping off the app out from memory and restarting it helps. 
Happens about once or twice a day with XiaoMi Redmi 9T, and less with Samsung A52.
I noticed it a few months ago, so it is not related to the latest version(s) of the Android app.

chuot, 2022-04-12:
This can happen if you are using Bluetooth headphones and connect/disconnect them while Rdio Scanner is in the background, but it would result in no link on the screen. This is also true for any application that attempts to steal audio focus from Rdio Scanner.

Is FFMPEG installed on your Windows PC where Rdio Scanner is? This would allow Rdio Scanner to convert mp3 files to m4a, which are smaller anyway but usually solve some audio issues on the native app side. I know FFMPEG is harder to install on Windows, but it would be nice to test it out and see if it fixes your problem.

I have a Samsung S9+ on which Rdio Scanner runs 24 hours a day and I have never had such an audio issue. So that won't help me understand what's going on with your own phones.

If possible, get a logcat with adb from your phone when this happens, that would help me a lot to solve this one.

Thanks

noamlivne, 2022-04-12:
Thank you for your response.
There is no issue with the Link. The Link is kept on all the time during this issue. The only issue is that the audio is gone.

I will disable the Bluetooth completely and see if that works.

I will try to install FFMPEG on my Windows. I am not sure when I'll get to it.
Same thing with ADB LOGCAT. I need to read about it and understand how to use it. And in addition, I wonder if the phone needs to be connected to the computer when it happens. It happens out of the blue.

Anyway, maximum I will learn to live with it as I I live with Android OS kicking the app to conserve battery from time to time.

chuot, 2022-04-25:
Closing for now, not enough information.

closed 2022-04-25 by chuot