chuot/rdio-scanner #164: (enhance)  ProScan Metadata

kcwebby, 2022-04-12:
When dirwatch'ing from a ProScan instance, the transference of meta data is limited to transfer into RDIO.

ProScan has a specific set of variables that can be added into the filename; where some are available in the mask of RDIO, but some are not.  My understanding is ProScan development is limited and not likely to change.   This software is used for physical scanner units.

Example screenshot of the fields available in ProScan for pickup into RDIO

[image]

chuot, 2022-05-10:
After looking at it, I'm still wondering which tag I should add. The current dirwatch metatags are:

```
#DATE - extract the date like 20201231 (YYYYMMMDD), 2020-12-31 (YYYY-MM-DD) or 2020_12_31 (YYYY_MM_DD).
#HZ - extract the frequency in hertz like 119100000.
#KHZ - extract the frequency in kilohertz like 119100.
#MHZ - extract the frequency in megahertz like 119.100.
#SYS - extract the system id like 11.
#TIME - extract the local time like 0853439 (HHMMSS), 08-34-39 (HH-MM-SS) or 08:34:39 (HH:MM:SS).
#TG - extract the talkgroup id like 54241.
#TGHZ - extract the frequency in hertz like 119100000 and set the talkgroup id to 119100.
#TGKHZ - extract the frequency in kilohertz like 119100.000 or 119100 and set the talkgroup id to 119100.
#TGMHZ - extract the frequency in megahertz like 119.100000 or 119 and set the talkgroup id to 119100.
#UNIT - extract the unit id like 4424001.
#ZTIME - extract the zulu time like 0453439, 08-34-39 or 04:34:39.
```

kcwebby, 2022-05-10:
This is what proscan is capable of, you can see the fileoutput that I am
currently using.

%D_%TT_%G__%FT_%U3_F
  05-10-22_17-37-01_TAMPA FIRE__11-061__F.mp3

[image: image.png]

chuot, 2022-05-11:
From your example, it looks like the %U3 is empty.

Can you post a couple of different filenames so I make sure I cover all scenarios?

chuot, 2022-05-26:
I guess I can close this issue as new metatags have been added since.

closed 2022-05-26 by chuot

kcwebby, 2022-05-26:
Yes thank you for working through this with me; I got lost and forgot to close this out since I was emailing directly the details.