chuot/rdio-scanner #166: m4a files from Trunk-Recorder not always working

kb2ear, 2022-04-13:
I tried to use the rdio-scanner plugin for the first time today. I noticed quite quickly that not all uploads were playing on rdio-scanner. Best I can figure is that there is something incompatible with the m4a files being generated by trunk-recorder but I have no idea why/what needs to be changed. 

To test the theory I used to upload-script to upload the m4a files and got the same result. But if I upload the wav files it always works. Maybe a switch in the plugin to upload the wav instead?

kb2ear, 2022-04-13:
trunk-recorder is doing this:
sox filename.wav --norm=-.01 -t wav - | fdkaac --silent  -p 2 --ignorelength -b 8000 -o filename.m4a -

kb2ear, 2022-04-13:
I may have been able to resolve by adding --moov-before-mdat to the fdkaac.

Thoughts?

kb2ear, 2022-04-14:
I did my testing in the webapp on windows running chrome. I guess I should test in the iphone app to see if it works there too.

audio conversion is enabled on the rdio-scanner instance since I was feeding it wav files from the uploadScript. ffmpeg is installed. 

So I'm guessing we need to get Luke to add the option in his code for the --moov-before-mdat for the m4a files he is generating.

chuot, 2022-04-25:
I tried to identify such m4a files which are not playing in the web or native app, but I couldn't find a single one on my own setup, which is mostly powered by trunk-recorder. I had such symptoms over 2 years ago and found out then that the `-movflags frag_keyframe+empty_moov` while converting with FFMPEG fixed it.

Please close the issue if you think it's something on the trunk-recorder side.

chuot, 2022-05-09:
Closing for now

closed 2022-05-09 by chuot