chuot/rdio-scanner #169: Is There a Way to Shrink the Database?

Dewey3, 2022-04-24:
I run Rdio-Scanner on its own Raspberry Pi 4 from a 128 GB micro SD card.  I was experimenting to see how many days I could hold on the 128 GB card since there was so much free space.  When I got somewhere north of 60 days, Windows File Explorer started showing the remaining free space (10.9 GB) in a red color due to the diminishing amount of free space.  I first set the prune option to 60 days, and when I noticed that I did not get any space back, I bumped it down to first 45 days, then down to 30 days.  Since no free space was reclaimed, I finally set the prune option to 1 day.  After waiting over a day, I noticed that the rdio-scanner.db is still the same size it was when Windows File Explorer started warning me about running out of space.  So my guess is that as the database grows, it does not automatically shrink when records are pruned.  With that said, is there a command that I can use on the rdio-scanner.db to reclaim the empty space of the records that have been pruned?  I know I can zero the database with the `>` (`$ > rdio-scanner.db`) command, but I don't want to have to rebuild the Rdio-Scanner configuration.  If all else fails, I'll just export out the configuration, zero the database, then reload the exported configuration.  I just wanted to see if there was a command line option I could use to shrink the database without going through the export/import.

chuot, 2022-04-25:
You can use the [VACUUM](https://www.sqlite.org/lang_vacuum.html) command when Rdio Scanner is stopped.

closed 2022-04-25 by chuot

Dewey3, 2022-04-25:
Thanks much Chrystian!  I considered VACUUM, but felt that would not work since I would need twice the space to start with... since my understanding of VACUUM is that it creates a duplicate database first before shinking the original, thus requiring twice the space.  I ended up exporting the configuration, creating a blank database with `$ > rdio-scanner.db`, then importing the exported configuration.  Everything went GREAT.  The only thing the export/import didn't do was restore my previous admin password from the default admin password, but that was certainly NO PROBLEM.  Thanks again Chrystian.