chuot/rdio-scanner #175: Archive Listening Catch Up To Live

Nokoa, 2022-05-05:
Currently while monitoring archives, playing one recording will prompt the next one to play. But it will only go on until it reaches the last file from when the user initially started playing archives. I would love to see a feature where archives would update to have the latest recordings, so playing archives would be able to catch up, and eventually be live.

chuot, 2022-05-09:
That is something feasible. At the end of the list, I could reload it to see if there is something new to continue the playback from there. I'm not sure however if flipping back to normal mode at the end is something everyone would like. In my case, I prefer the playback to stop instead of going live. But that's something I could and as an option. I will try to fit that in the v6.3.

Nokoa, 2022-05-09:
I was monitoring archives for an ongoing incident from the beginning, and as I was listening to archives, more and more new transmissions were coming in. However it eventually reached the last file in the archives based on when I initially started playing the archives, so I would have to click reset, and navigate back to the time when it left of to continue until it stops again. Perhaps continuing playing indefinitely may not be the right approach. Maybe the archives can refresh, or have a button to refresh them?

chuot, 2022-05-09:
When the playback mode has reached the end of the list, you can go to page 2, then back to page 1. This will refresh the list.

chuot, 2022-05-20:
The feature is now fully available in v6.3.1 where it will catch up until there are no more new calls in the archive according to search filters and then switch from playback mode (orange) to livefeed mode (green) and continue normal operation.

I must say that I have been trying this new feature for a few days now and I really like it! Honestly, this feature should have been around for a while.

Thanks for suggesting it!

Nokoa, 2022-05-29:
Thank you for working hard to implement this feature, I appreciate it. I have been using it for a couple days now. I love that playing archives will continue playing even the recordings that weren't present at the time I originally started playing. This helps in monitoring an ongoing incident from its start to present. However I feel that switching Rdio-Scanner actually to live once all archives have been played, might not be the correct approach. I have had instances, where I would multiple times a day check special announcement channels, for priority incidents, and sometimes there will be one or two recordings, I would listen to them, and then move over to the next announcement channel to do the same. However, because the new feature where it will go live after it finished playing the last recording available it will go live, it causes a bit of an unexpected, and unnatural flow of things. Because now it's playing my full scan list, even though I am only monitoring the one talkgroup. 

I know that this is slightly different from the regular operation of rdio-scanner, and might not fit. But I believe that perhaps the correct approach would be to say for example. I am checking recordings in the archives for talkgroup 2600. Once I have reached the last recording available for 2600, and if I am still on the archive page for that specific talkgroup or maybe even if I got back to the main screen, but haven't stopped, or changed the page of the archive I was selected on. If a new recording for 2600 only comes in, it would appear in the top of the archive list (depending on sort order), and play that. So rdio-scanner will remain in the yellow light playing mode. It will continue to play based on the archive parameters only. So if I change the talkgroup selected in the archives, the playing would stop, as it currently does while monitoring archives. Or if I toggle the yellow live feed button back to off. Or if I press pause.

Does this make sense?

chuot, 2022-05-30:
Well, I can go back to the old way like in v6.3.0. Switching to LIVE FEED mode was added in v6.3.1. I can revert back this change so that it disables the LIVE FEED (from yellow to red) when the playlist is completed. When you reach the end of the list, it automatically refreshes once to ensure that no new calls have been ingested since. If so, the playback mode continues to the end of the (refreshed) list and refreshes it again. If there are no new calls, LIVE FEED will switch from play mode (yellow) to offline mode (red).

I could also make this behavior configurable in server options, as in some cases I prefer to switch back to LIVE FEED mode from playback mode where my selected talkgroups will play as usual. In your case, you can also make sure that your selected talgroups on the SELECT TG panel match your search filters on the SEARCH CALL panel.

chuot, 2022-05-31:
The option has been added to v6.4.0.

closed 2022-06-27 by chuot

Nokoa, 2022-07-27:
Thank you for implementing. Works great!