chuot/rdio-scanner #176: Feature Request: Listener Timeout

Dewey3, 2022-05-08:
Just a thought... what about a listener timeout in the Access portion of Rdio-Scanner so a listener cannot hog the system?  This may be a lot of work since it would have to track each listener (? by IP address ?) instead of anyone and everyone using that access code, so not worth the effort... I understand.

chuot, 2022-05-09:
I'm not sure I fully understand your request. On what exactly would you like users to timed out on?

There're already timeout mechanisms to close a client connection if a websocket is opened without completing the registration process (for ports scanner harvesting the Internet), client connections with the same access code that is expired or exceeding the limit.

If you're seeing spikes in the listeners count, that may be because of ports scanner, but the count should dropped after 10 seconds or so if these scanners don't complete the Rdio Scanner registration process.

chuot, 2022-05-26:
You're asking for a demonstration mode, kinda.

What we do on our corporate website is showing an iframe which we close after 10 minutes with some javascript. Then you can configure your webserver to check the referer and only allow your own domain. I'ts not bullet proof but it works well enough.

I'm closing this issue as we have no plan to add such demo mode for now.

closed 2022-05-26 by chuot