chuot/rdio-scanner #177: Bug: 6.2.5 - Random Stopping

JayManCanada6014, 2022-05-12:
I have Rdio running on a VPS  4Cpu. 8 ram and 200GB SSD

For the past month or so, Rdio will just stop and won't ingest calls until I restart it,   I thought maybe it was a DB pruning again, but the logs don't show that happening anymore.  508,000+ records and I only keep them for 30days.

There is a increase in CPU usage when it happens.  
[image]

I have a screen showing  journalctl -u rdio-scanner.service -f  always so i just grabbed this from that.

May 12 16:26:01 rdio-scanner rdio-scanner[584269]: 2022/05/12 16:26:01 newcall: system=100 talkgroup=8960 file=896                                                                                                                      0-1652387152_142125000-call_10354.m4a success
May 12 16:26:06 rdio-scanner rdio-scanner[584269]: 2022/05/12 16:26:06 newcall: system=2000 talkgroup=1 file=16523                                                                                                                      87164-1.m4a success
May 12 16:26:06 rdio-scanner rdio-scanner[584269]: 2022/05/12 16:26:06 newcall: system=999 talkgroup=1 file=165238                                                                                                                      7166-1.m4a success
May 12 16:26:09 rdio-scanner rdio-scanner[584269]: 2022/05/12 16:26:09 newcall: system=100 talkgroup=8960 file=896                                                                                                                      0-1652387158_142950000-call_10356.m4a success
^C
root@rdio-scanner:~# service rdio-scanner restart
root@rdio-scanner:~# journalctl -u rdio-scanner.service -f
-- Logs begin at Sun 2021-12-19 13:43:46 EST. --
May 12 16:51:51 rdio-scanner systemd[1]: rdio-scanner.service: Succeeded.
May 12 16:51:51 rdio-scanner systemd[1]: Stopped The perfect software-defined radio companion.
May 12 16:51:51 rdio-scanner systemd[1]: Started The perfect software-defined radio companion.
May 12 16:51:51 rdio-scanner rdio-scanner[835269]: Rdio Scanner v6.2.5
May 12 16:51:51 rdio-scanner rdio-scanner[835269]: ----------------------------------
May 12 16:51:51 rdio-scanner rdio-scanner[835269]: 2022/05/12 16:51:51 server started
May 12 16:51:51 rdio-scanner rdio-scanner[835269]: 2022/05/12 16:51:51 base folder is /opt/v6
May 12 16:51:51 rdio-scanner rdio-scanner[835269]: 2022/05/12 16:51:51 main interface at http://rdio-scanner:3001
May 12 16:51:51 rdio-scanner rdio-scanner[835269]: 2022/05/12 16:51:51 main interface at https://rdio-scanner
May 12 16:51:51 rdio-scanner rdio-scanner[835269]: 2022/05/12 16:51:51 admin interface at https://rdio-scanner/adm                                                                                                                      in
May 12 16:51:53 rdio-scanner rdio-scanner[835269]: 2022/05/12 16:51:53 newcall: system=3 talkgroup=3 file=EMSPAGIN                                                                                                                      G-3_20220512_16-45-38.m4a success
May 12 16:51:53 rdio-scanner rdio-scanner[835269]: 2022/05/12 16:51:53 newcall: system=3 talkgroup=3 file=EMSPAGIN                                                                                                                      G-3_20220512_16-46-34.m4a success
May 12 16:51:56 rdio-scanner rdio-scanner[835269]: 2022/05/12 16:51:56 listeners count is 17
May 12 16:51:58 rdio-scanner rdio-scanner[835269]: 2022/05/12 16:51:58 listeners count is 19
May 12 16:51:59 rdio-scanner rdio-scanner[835269]: 2022/05/12 16:51:59 newcall: system=102 talkgroup=35232 file=35                                                                                                                      232-1652388706_142995000-call_23268.m4a success
May 12 16:52:01 rdio-scanner rdio-scanner[835269]: 2022/05/12 16:52:01 newcall: system=201 talkgroup=1503 file=150                                                                                                                      3-1652388712_867637500-call_31151.m4a success
May 12 16:52:04 rdio-scanner rdio-scanner[835269]: 2022/05/12 16:52:04 newcall: system=102 talkgroup=35456 file=35                                                                                                                      456-1652388713_141375000-call_23269.m4a success
May 12 16:52:06 rdio-scanner rdio-scanner[835269]: 2022/05/12 16:52:06 newcall: system=102 talkgroup=35440 file=35  

As you can see, the listeners are still there (green line) just no new calls.

[image]

chuot, 2022-05-13:
The next time this happens, `kill -3` the process, grab all the output and post them back.

Thanks

JayManCanada6014, 2022-05-13:
[Rdio-Scanner Logs.txt](https://github.com/chuot/rdio-scanner/files/8691656/Rdio-Scanner.Logs.txt)
Also it was 24hours since the last one, almost to the minute. 

Sorry for the formatting, there terminal I grabbed it messed it up a bit.

chuot, 2022-05-14:
I see from your logs that it kinda hangs on getting a new file descriptor.

Did you get any too many opened files error in your logs?

It might be worth boosting some values in /etc/security/limits.conf like [here](https://github.com/chuot/rdio-scanner/discussions/178). I'd go even higher in your case.

I've also stress tested an instance where I loaded in 32000 new calls in 82 minutes, while doing database pruning every 10 seconds on fedora 36. It never hanged or so.

My demo instance (docker) and my main instance (linux-amd64) are running for more than a month and never hanged. I'm starting to wonder if Fedora do have some kind of a secret sauce that Ubuntu doesn't...

JayManCanada6014, 2022-05-15:
> 

When I restarted it again today,  it was saying there was too many open files from the /watch directory I have setup.  I went in there and there was like 500 files Rdio hasn't been deleting on ingest.   I've changed mine to the same as the other posting you linked to and lets see what happens.  Fedora/CentOS may just handle it batter who knows.

JayManCanada6014, 2022-05-15:
> When I restarted it again today, it was saying there was too many open files from the /watch directory I have setup. I went in there and there was like 500 files Rdio hasn't been deleting on ingest. [...]
[dump.log](https://github.com/chuot/rdio-scanner/files/8695624/dump.log)

Well that didn't work,  locked up 3 times today so far.  Attached is new log with some better formatting.

JayManCanada6014, 2022-05-15:
[dump.log](https://github.com/chuot/rdio-scanner/files/8695625/dump.log)

chuot, 2022-05-15:
I can't look at the logs right now. I will look at it later.

You should reboot the box when you hit too many files error.

What values you've put in /etc/security/limits.conf ?

JayManCanada6014, 2022-05-15:
Rebooted already this morning
* soft nofile 65536
* hard nofile 65536
As suggested in the post,  I've rebooted now again.

chuot, 2022-05-15:
Change 65536 to 1000000

JayManCanada6014, 2022-05-16:
1000000  didn't fix it.
I even wiped my DB to see if maybe that is the cause and still the same. It seems to be getting worse, I have cron restarting the service now once an hour.

I'm tempted to just change to a different OS at this point.

chuot, 2022-05-16:
Wait before doing so.

Have tou tried running the server at the command line without using systemd ad proposed on discord?

It must be related to ulimit. Make sure ulimit -n gives the proper value before running the server at the command line.

I'm still testing and will come up with something before the end of the day.

JayManCanada6014, 2022-05-16:
> Wait before doing so.
>
> Have tou tried running the server at the command line without using systemd ad proposed on discord?
>
> It must be related to ulimit. Make sure ulimit -n gives the proper value [...]

Someone suggested the same think in Trunk Recorder discord   so I am running it not as a service now.  
root@rdio-scanner:/opt/v6# ulimit -n
1024

chuot, 2022-05-16:
Ok, if you are still getting 1024 when doing `ulimit -n`, that means your mods in `/etc/security/limits.conf` didn't work. I would review your changes to make sure they're effective.

I tested again this morning with 100 concurrent users and running a script that loop 16000 times to load random sys/tg through a dirwatch and through the api for a total of 32000 new calls in 81 minutes. Audio conversion and duplicate check were also enabled. Database pruning was set to every 10 seconds during that stress test. The client and the administrative dashboard were responding to all requests without an issue. I was also monitoring with `lsof` to count the number of open file descriptors during the whole process and it was stable at 145. Honnestly, I can't see how I can stress more a single instance of Rdio Scanner!

Clearly your issue is the limit in the file descriptors count and something else than Rdio Scanner is eating them up. Note that network connections also consume file descriptors. 100 or the 145 were the 100 connected clients, the rest were linked to normal operations (api, dirwatch, database, ...)

One more thing, all these tests where done with the upcoming v6.3 where I change some codes about GO channel sizes and idle timeout on SQLite connections. It was also compile with the latest GO 1.18.2 and some libraries like fsnotify and sqlite were updated too. These newer versions may also have fixed something in regards to leaking file descriptors. Not sure tho.

When this will be sorted out, I will sure add something about ulimits to the FAQ section of the PDF manual.

JayManCanada6014, 2022-05-16:
So far running it as a standalone it's been ok for 2 hours.  Will keep monitoring it.
Funny never had issues at all with it until last week,  something got patched in Ubuntu maybe

chuot, 2022-05-18:
Would it be possible to deactivate any dirwatch for a full day to see if it still does it around midnight?

JayManCanada6014, 2022-05-18:
> Would it be possible to deactivate any dirwatch for a full day to see if it still does it around midnight?

Dir Watch is disabled as of now.   
You think it's I am using Proscan and having a script upload recording to Rdio that is breaking it.   I've been doing it this way for a close to a year.

chuot, 2022-05-18:
I'm in the process of rewriting some code in the dirwatch processor. I've identified a couple of things I'm not doing properly according to the [fsnotify](https://github.com/fsnotify/fsnotify) documentation.

I will publish a test version when it will be ready. Windows users are also plagued by this issue.

JayManCanada6014, 2022-05-19:
> I'm in the process of rewriting some code in the dirwatch processor. I've identified a couple of things I'm not doing properly according to the [fsnotify](https://github.com/fsnotify/fsnotify) [...]

I think you have the cause.  Nothing at midnight last night.

[image]

chuot, 2022-05-19:
I need to test a bit more before releasing v6.3. Hopefully this will fix the problem once and for all!

Thanks again for providing evidence.

JayManCanada6014, 2022-05-19:
> I need to test a bit more before releasing v6.3. Hopefully this will fix the problem once and for all!
> 
> Thanks again for providing evidence.

Just happened again with Dirwatch still disabled.   I didn't remove it from the configs but I did turn it off.

JayManCanada6014, 2022-05-19:
6.3.0 now online as of 3pm with dirwatch turned on, lets see what happens.

JayManCanada6014, 2022-05-19:
And stopped again at 3:50pm    Oddly enough kill -3 didn't produce any logs

JayManCanada6014, 2022-05-19:
Yeah still going on, I even removed tail scale just as a common thing between the installs and I wasn't even using it to upload calls and as I was writing this it stopped again.   I've also opened a ticket with my VPS company as well.  Honestly I am out of ideas.

chuot, 2022-05-19:
Can you try on Fedora 36 (or 35) Server Edition?

Mine (v6.3 on F35 barebone) has been steady for more than a day with both api and dirwatch. Even running trunk-recorder and rtl-airband on the same box!

I'm wondering if it has something to do with libraries?

JayManCanada6014, 2022-05-19:
> Can you try on Fedora 36 (or 35) Server Edition?
> 
> Mine (v6.3 on F35 barebone) has been steady for more than a day with both api and dirwatch. Even running trunk-recorder and rtl-airband on the same box!
> 
> I'm wondering if it has something to do with libraries?

We have investigated the issue, our technical department has made certain adjustments to the server, please let us know if the problem continues, and if so - please get back to us with root login and password so that our technician can access the server.

If this doesn't fix it I will reinstall with Fedora 35 which is the only Fedora they have for install.

JayManCanada6014, 2022-05-20:
It didn't fix it, will try fedora tomorrow, also have the ticket open with the provider.

chuot, 2022-05-20:
Hey, my demo instance was frozen this morning! It is using the docker image and runing with podman. No dirwatch, just using API and feed by my main instance.

I was able to get a good log output with `kill -3`. There's a bunch of GO routines in `semacquire` state, which means they are all waiting for a mutex to unlock. I'm in the process of reviewing those logs line by line and referencing the source code. It will take some times, but at least I will be able to isolate the root cause of this dead lock.

More to come...

chuot, 2022-05-20:
Ok, the mutex on Clients.Count() was the culprit. Please try with v6.3.1 which is just out.

JayManCanada6014, 2022-05-20:
> Ok, the mutex on Clients.Count() was the culprit. Please try with v6.3.1 which is just out.

Running as of 9am, we shall see what happens.

JayManCanada6014, 2022-05-20:
> Ok, the mutex on Clients.Count() was the culprit. Please try with v6.3.1 which is just out.

Lasted until 2:11pm and stopped again.

chuot, 2022-05-20:
I need the kill -3 output

JayManCanada6014, 2022-05-20:
> I need the kill -3 output

Attached. 
[dump.log](https://github.com/chuot/rdio-scanner/files/8743658/dump.log)

chuot, 2022-05-20:
Can you try to deactivate the listener count from the config / options ?

JayManCanada6014, 2022-05-20:
> Can you try to deactivate the listener count from the config / options ?

Done as of 3:53pm  Will reply when I see anything of interest.

chuot, 2022-05-20:
Keep posting the kill -3 outputs when it froze

JayManCanada6014, 2022-05-20:
> Keep posting the kill -3 outputs when it froze

Died at 4:15pm, but I was outside and just restarted it on my phone.  I'll grab the dump next one. 
It does seem to happen more when there are people online,  overnights works fine very few on but as people login during the day it happens more.
Am I getting hit from the outside?

JayManCanada6014, 2022-05-25:
I think we are good now with 6.3.3

closed 2022-05-26 by chuot