chuot/rdio-scanner #181: CPU stuck at 100% on VM

tdduarte, 2022-05-17:
I noticed running the latest on vultr on Ubuntu and Debian after a period of time of time the CPU pegs at 100%... Any ideas what maybe wrong?  Is running as a service..

chuot, 2022-05-18:
It may be related to issue #177.

tdduarte, 2022-05-18:
Ya seems similar... also not sure if this helps but my setup is API only... no directory watching..

tdduarte, 2022-05-18:
It actually crashed some point over night or this morning... but way after the cpu was stuck at 100%... it also stops the trunk recorders and when rdio comes back up the truck records flood the system with old messages

chuot, 2022-05-19:
if you can provide journal logs from when it crashes that would be helpful.

Also make sure your VM has enough RAM, disk space and the `ulimit -Hn` for the user running Rdio Scanner is above 60000.

tdduarte, 2022-05-20:
Looks like it just stops.. CPU is still stuck at 100% but website is no longer responding.. worked for about 24 hours..

root@scanner:~# ulimit -Hn
1048576

May 19 22:17:01 scanner CRON[3757]: pam_unix(cron:session): session closed for user root
May 19 22:17:01 scanner CRON[3758]: (root) CMD (   cd / && run-parts --report /etc/cron.hourly)
May 19 22:17:01 scanner CRON[3757]: pam_unix(cron:session): session opened for user root(uid=0) by (uid=0)
May 19 21:53:44 scanner rdio-scanner[1936]: 2022/05/19 21:53:44 newcall: system=5 talkgroup=33968 file=33968-1652996953_855662500-call_46490.m4a success
May 19 21:53:43 scanner rdio-scanner[1936]: 2022/05/19 21:53:43 newcall: system=5 talkgroup=33712 file=33712-1652996958_855662500-call_46491.m4a success
May 19 21:53:43 scanner rdio-scanner[1936]: 2022/05/19 21:53:43 newcall: system=5 talkgroup=36752 file=36752-1652996998_855662500-call_46501.m4a success
May 19 21:53:42 scanner rdio-scanner[1936]: 2022/05/19 21:53:42 newcall: system=5 talkgroup=33264 file=33264-1652996978_855662500-call_46495.m4a success
May 19 21:53:42 scanner rdio-scanner[1936]: 2022/05/19 21:53:42 newcall: system=5 talkgroup=33264 file=33264-1652996947_855662500-call_46488.m4a success
May 19 21:53:40 scanner rdio-scanner[1936]: 2022/05/19 21:53:40 newcall: system=5 talkgroup=36752 file=36752-1652996985_855662500-call_46497.m4a success
May 19 21:53:40 scanner rdio-scanner[1936]: 2022/05/19 21:53:40 newcall: system=5 talkgroup=33968 file=33968-1652996964_855662500-call_46493.m4a success
May 19 21:53:34 scanner rdio-scanner[1936]: 2022/05/19 21:53:34 newcall: system=10 talkgroup=1 file=1652997211-1.mp3 success
May 19 21:53:33 scanner rdio-scanner[1936]: 2022/05/19 21:53:33 newcall: system=1 talkgroup=5 file=5-1652997188_483525000-call_4.wav success
May 19 21:53:30 scanner rdio-scanner[1936]: 2022/05/19 21:53:30 newcall: system=5 talkgroup=36752 file=36752-1652997035_855662500-call_46509.m4a success

tdduarte, 2022-05-20:
kill -3

`May 19 21:07:47 scanner rdio-scanner[1936]: main.(*Clients).Count(0x90d6260)
May 19 21:07:47 scanner rdio-scanner[1936]:         /home/chuot/Dev/rdio-scanner/server/client.go:212 +0x44
May 19 21:07:47 scanner rdio-scanner[1936]: main.(*Client).Init(0x9e14280, 0x90808a0, 0x9c5d8c0)
May 19 21:07:47 scanner rdio-scanner[1936]:         /home/chuot/Dev/rdio-scanner/server/client.go:52 +0x34
May 19 21:07:47 scanner rdio-scanner[1936]: main.main.func1({0x88e9d24, 0x1e5d1c20}, 0xa243280)
May 19 21:07:47 scanner rdio-scanner[1936]:         /home/chuot/Dev/rdio-scanner/server/main.go:139 +0x2fd
May 19 21:07:47 scanner rdio-scanner[1936]: net/http.HandlerFunc.ServeHTTP(0x900dd98, {0x88e9d24, 0x1e5d1c20}, 0xa243280)
May 19 21:07:47 scanner rdio-scanner[1936]:         /home/chuot/go/src/net/http/server.go:2084 +0x34
May 19 21:07:47 scanner rdio-scanner[1936]: net/http.(*ServeMux).ServeHTTP(0x8ca46e0, {0x88e9d24, 0x1e5d1c20}, 0xa243280)
May 19 21:07:47 scanner rdio-scanner[1936]:         /home/chuot/go/src/net/http/server.go:2462 +0x155
May 19 21:07:47 scanner rdio-scanner[1936]: net/http.serverHandler.ServeHTTP({0x905eea0}, {0x88e9d24, 0x1e5d1c20}, 0xa243280)
May 19 21:07:47 scanner rdio-scanner[1936]:         /home/chuot/go/src/net/http/server.go:2916 +0x371
May 19 21:07:47 scanner rdio-scanner[1936]: net/http.(*conn).serve(0x9d9ae40, {0x88e9ef0, 0x9015ab8})
May 19 21:07:47 scanner rdio-scanner[1936]:         /home/chuot/go/src/net/http/server.go:1966 +0x67d
May 19 21:07:47 scanner rdio-scanner[1936]: created by net/http.(*Server).Serve
May 19 21:07:47 scanner rdio-scanner[1936]:         /home/chuot/go/src/net/http/server.go:3071 +0x498
May 19 21:07:47 scanner rdio-scanner[1936]: goroutine 17498 [chan send, 50 minutes]:
May 19 21:07:47 scanner rdio-scanner[1936]: main.(*Api).HandleCall(0x900c9d8, {0x18defaa0, 0x24}, 0x1e5d1d60, {0x88e9d24, 0x1e5d1cc0})
May 19 21:07:47 scanner rdio-scanner[1936]:         /home/chuot/Dev/rdio-scanner/server/api.go:97 +0x149
May 19 21:07:47 scanner rdio-scanner[1936]: main.(*Api).CallUploadHandler(0x900c9d8, {0x88e9d24, 0x1e5d1cc0}, 0xa243300)
May 19 21:07:47 scanner rdio-scanner[1936]:         /home/chuot/Dev/rdio-scanner/server/api.go:80 +0x566
May 19 21:07:47 scanner rdio-scanner[1936]: net/http.HandlerFunc.ServeHTTP(0x900dd88, {0x88e9d24, 0x1e5d1cc0}, 0xa243300)
May 19 21:07:47 scanner rdio-scanner[1936]:         /home/chuot/go/src/net/http/server.go:2084 +0x34
May 19 21:07:47 scanner rdio-scanner[1936]: net/http.(*ServeMux).ServeHTTP(0x8ca46e0, {0x88e9d24, 0x1e5d1cc0}, 0xa243300)
May 19 21:07:47 scanner rdio-scanner[1936]:         /home/chuot/go/src/net/http/server.go:2462 +0x155
May 19 21:07:47 scanner rdio-scanner[1936]: net/http.serverHandler.ServeHTTP({0x905eea0}, {0x88e9d24, 0x1e5d1cc0}, 0xa243300)
May 19 21:07:47 scanner rdio-scanner[1936]:         /home/chuot/go/src/net/http/server.go:2916 +0x371
May 19 21:07:47 scanner rdio-scanner[1936]: net/http.(*conn).serve(0x9d9aea0, {0x88e9ef0, 0x9015ab8})
May 19 21:07:47 scanner rdio-scanner[1936]:         /home/chuot/go/src/net/http/server.go:1966 +0x67d
May 19 21:07:47 scanner rdio-scanner[1936]: created by net/http.(*Server).Serve
May 19 21:07:47 scanner rdio-scanner[1936]:         /home/chuot/go/src/net/http/server.go:3071 +0x498
May 19 21:07:47 scanner rdio-scanner[1936]: goroutine 17499 [IO wait, 50 minutes]:
May 19 21:07:47 scanner rdio-scanner[1936]: internal/poll.runtime_pollWait(0xe6dedf4c, 0x72)
May 19 21:07:47 scanner rdio-scanner[1936]:         /home/chuot/go/src/runtime/netpoll.go:302 +0x52
May 19 21:07:47 scanner rdio-scanner[1936]: internal/poll.(*pollDesc).wait(0x24b5d004, 0x72, 0x0)
May 19 21:07:47 scanner rdio-scanner[1936]:         /home/chuot/go/src/internal/poll/fd_poll_runtime.go:83 +0x37
May 19 21:07:47 scanner rdio-scanner[1936]: internal/poll.(*pollDesc).waitRead(...)
May 19 21:07:47 scanner rdio-scanner[1936]:         /home/chuot/go/src/internal/poll/fd_poll_runtime.go:88
May 19 21:07:47 scanner rdio-scanner[1936]: internal/poll.(*FD).Read(0x24b5cff0, {0x9e5602d, 0x1, 0x1})
May 19 21:07:47 scanner rdio-scanner[1936]:         /home/chuot/go/src/internal/poll/fd_unix.go:167 +0x1e0
May 19 21:07:47 scanner rdio-scanner[1936]: net.(*netFD).Read(0x24b5cff0, {0x9e5602d, 0x1, 0x1})
May 19 21:07:47 scanner rdio-scanner[1936]:         /home/chuot/go/src/net/fd_posix.go:55 +0x3f
May 19 21:07:47 scanner rdio-scanner[1936]: net.(*conn).Read(0x170e38a0, {0x9e5602d, 0x1, 0x1})
May 19 21:07:47 scanner rdio-scanner[1936]:         /home/chuot/go/src/net/net.go:183 +0x4f
May 19 21:07:47 scanner rdio-scanner[1936]: net/http.(*connReader).backgroundRead(0x9e56020)
May 19 21:07:47 scanner rdio-scanner[1936]:         /home/chuot/go/src/net/http/server.go:672 +0x46
May 19 21:07:47 scanner rdio-scanner[1936]: created by net/http.(*connReader).startBackgroundRead
May 19 21:07:47 scanner rdio-scanner[1936]:         /home/chuot/go/src/net/http/server.go:668 +0xd6
May 19 21:07:47 scanner rdio-scanner[1936]: goroutine 17500 [chan send, 49 minutes]:
May 19 21:07:47 scanner rdio-scanner[1936]: main.(*Api).HandleCall(0x900c9d8, {0x13194840, 0x24}, 0x1e5d1ea0, {0x88e9d24, 0x1e5d1e00})
May 19 21:07:47 scanner rdio-scanner[1936]:         /home/chuot/Dev/rdio-scanner/server/api.go:97 +0x149
May 19 21:07:47 scanner rdio-scanner[1936]: main.(*Api).CallUploadHandler(0x900c9d8, {0x88e9d24, 0x1e5d1e00}, 0xa243380)
May 19 21:07:47 scanner rdio-scanner[1936]:         /home/chuot/Dev/rdio-scanner/server/api.go:80 +0x566
[... cut ...]
May 19 21:07:48 scanner rdio-scanner[1936]: net/http.(*conn).serve(0x9ee8780, {0x88e9ef0, 0x9015ab8})
May 19 21:07:48 scanner rdio-scanner[1936]:         /home/chuot/go/src/net/http/server.go:1966 +0x67d
May 19 21:07:48 scanner rdio-scanner[1936]: created by net/http.(*Server).Serve
May 19 21:07:48 scanner rdio-scanner[1936]:         /home/chuot/go/src/net/http/server.go:3071 +0x498
May 19 21:07:48 scanner rdio-scanner[1936]: goroutine 17631 [chan send, 36 minutes]:
May 19 21:07:48 scanner rdio-scanner[1936]: main.(*Api).HandleCall(0x900c9d8, {0xd4e0b10, 0x24}, 0x9f1ed20, {0x88e9d2

tdduarte, 2022-05-20:
just upgraded to 6.3.. see what happens

chuot, 2022-05-20:
Same thing as isse #177, deadlock on Clients.Count which you can see in your logs above, a lot of [semacquire].

Try with v6.3.1 which is ready to download.

tdduarte, 2022-05-20:
Great!!! I will give it a try... Thank you for your help on this!

tdduarte, 2022-05-21:
We just crashed not even 20 mins after loading 6.3.1

May 21 03:47:55 scanner rdio-scanner[798]: main.(*Api).HandleCall(0xa80c9d8, {0xb7a4c60, 0x24}, 0xa8f72c0, {0x88eadcc, 0xa8f6fa0})
May 21 03:47:55 scanner rdio-scanner[798]: goroutine 928 [chan send, 32 minutes]:
May 21 03:47:55 scanner rdio-scanner[798]:         /home/chuot/go/src/net/http/server.go:668 +0xd6
May 21 03:47:55 scanner rdio-scanner[798]: created by net/http.(*connReader).startBackgroundRead
May 21 03:47:55 scanner rdio-scanner[798]:         /home/chuot/go/src/net/http/server.go:672 +0x46
May 21 03:47:55 scanner rdio-scanner[798]: net/http.(*connReader).backgroundRead(0xa9620a0)
May 21 03:47:55 scanner rdio-scanner[798]:         /home/chuot/go/src/net/net.go:183 +0x4f
May 21 03:47:55 scanner rdio-scanner[798]: net.(*conn).Read(0xb649910, {0xa9620ad, 0x1, 0x1})
May 21 03:47:55 scanner rdio-scanner[798]:         /home/chuot/go/src/net/fd_posix.go:55 +0x3f
May 21 03:47:55 scanner rdio-scanner[798]: net.(*netFD).Read(0xa9260f0, {0xa9620ad, 0x1, 0x1})
May 21 03:47:55 scanner rdio-scanner[798]:         /home/chuot/go/src/internal/poll/fd_unix.go:167 +0x1e0
May 21 03:47:55 scanner rdio-scanner[798]: internal/poll.(*FD).Read(0xa9260f0, {0xa9620ad, 0x1, 0x1})
May 21 03:47:55 scanner rdio-scanner[798]:         /home/chuot/go/src/internal/poll/fd_poll_runtime.go:88
May 21 03:47:55 scanner rdio-scanner[798]: internal/poll.(*pollDesc).waitRead(...)
May 21 03:47:55 scanner rdio-scanner[798]:         /home/chuot/go/src/internal/poll/fd_poll_runtime.go:83 +0x37
May 21 03:47:55 scanner rdio-scanner[798]: internal/poll.(*pollDesc).wait(0xa926104, 0x72, 0x0)
May 21 03:47:55 scanner rdio-scanner[798]:         /home/chuot/go/src/runtime/netpoll.go:302 +0x52
May 21 03:47:55 scanner rdio-scanner[798]: internal/poll.runtime_pollWait(0xe7ce96cc, 0x72)
May 21 03:47:55 scanner rdio-scanner[798]: goroutine 930 [IO wait, 32 minutes]:
May 21 03:47:55 scanner rdio-scanner[798]:         /home/chuot/go/src/net/http/server.go:668 +0xd6
May 21 03:47:55 scanner rdio-scanner[798]: created by net/http.(*connReader).startBackgroundRead
May 21 03:47:55 scanner rdio-scanner[798]:         /home/chuot/go/src/net/http/server.go:672 +0x46
May 21 03:47:55 scanner rdio-scanner[798]: net/http.(*connReader).backgroundRead(0xa962040)
May 21 03:47:55 scanner rdio-scanner[798]:         /home/chuot/go/src/net/net.go:183 +0x4f
May 21 03:47:55 scanner rdio-scanner[798]: net.(*conn).Read(0xb6487f8, {0xa96204d, 0x1, 0x1})
May 21 03:47:55 scanner rdio-scanner[798]:         /home/chuot/go/src/net/fd_posix.go:55 +0x3f
May 21 03:47:55 scanner rdio-scanner[798]: net.(*netFD).Read(0xa86ea00, {0xa96204d, 0x1, 0x1})
May 21 03:47:55 scanner rdio-scanner[798]:         /home/chuot/go/src/internal/poll/fd_unix.go:167 +0x1e0
May 21 03:47:55 scanner rdio-scanner[798]: internal/poll.(*FD).Read(0xa86ea00, {0xa96204d, 0x1, 0x1})
May 21 03:47:55 scanner rdio-scanner[798]:         /home/chuot/go/src/internal/poll/fd_poll_runtime.go:88
May 21 03:47:55 scanner rdio-scanner[798]: internal/poll.(*pollDesc).waitRead(...)
May 21 03:47:55 scanner rdio-scanner[798]:         /home/chuot/go/src/internal/poll/fd_poll_runtime.go:83 +0x37
May 21 03:47:55 scanner rdio-scanner[798]: internal/poll.(*pollDesc).wait(0xa86ea14, 0x72, 0x0)
May 21 03:47:55 scanner rdio-scanner[798]:         /home/chuot/go/src/runtime/netpoll.go:302 +0x52
May 21 03:47:55 scanner rdio-scanner[798]: internal/poll.runtime_pollWait(0xe7ce9eb8, 0x72)
May 21 03:47:55 scanner rdio-scanner[798]: goroutine 932 [IO wait, 32 minutes]:
May 21 03:47:55 scanner rdio-scanner[798]:         /home/chuot/go/src/net/http/server.go:3071 +0x498
May 21 03:47:55 scanner rdio-scanner[798]: created by net/http.(*Server).Serve
May 21 03:47:55 scanner rdio-scanner[798]:         /home/chuot/go/src/net/http/server.go:1966 +0x67d
May 21 03:47:55 scanner rdio-scanner[798]: net/http.(*conn).serve(0xa880540, {0x88eaf98, 0xa815ad0})
May 21 03:47:55 scanner rdio-scanner[798]:         /home/chuot/go/src/net/http/server.go:2916 +0x371
May 21 03:47:55 scanner rdio-scanner[798]: net/http.serverHandler.ServeHTTP({0xa85eea0}, {0x88eadcc, 0xa8f6000}, 0xa888a00)
May 21 03:47:55 scanner rdio-scanner[798]:         /home/chuot/go/src/net/http/server.go:2462 +0x155
May 21 03:47:55 scanner rdio-scanner[798]: net/http.(*ServeMux).ServeHTTP(0x8ca6700, {0x88eadcc, 0xa8f6000}, 0xa888a00)
May 21 03:47:55 scanner rdio-scanner[798]:         /home/chuot/go/src/net/http/server.go:2084 +0x34
May 21 03:47:55 scanner rdio-scanner[798]: net/http.HandlerFunc.ServeHTTP(0xa80dd80, {0x88eadcc, 0xa8f6000}, 0xa888a00)
May 21 03:47:55 scanner rdio-scanner[798]:         /home/chuot/Dev/rdio-scanner/server/api.go:59 +0x375
May 21 03:47:55 scanner rdio-scanner[798]: main.(*Api).CallUploadHandler(0xa80c9d8, {0x88eadcc, 0xa8f6000}, 0xa888a00)
May 21 03:47:55 scanner rdio-scanner[798]:         /home/chuot/go/src/mime/multipart/multipart.go:314
May 21 03:47:55 scanner rdio-scanner[798]: mime/multipart.(*Reader).NextPart(...)
May 21 03:47:55 scanner rdio-scanner[798]:         /home/chuot/go/src/mime/multipart/multipart.go:346 +0x45b
May 21 03:47:55 scanner rdio-scanner[798]: mime/multipart.(*Reader).nextPart(0xa9b2240, 0x0)
May 21 03:47:55 scanner rdio-scanner[798]:         /home/chuot/go/src/fmt/errors.go:24 +0x111
May 21 03:47:55 scanner rdio-scanner[798]: fmt.Errorf({0x86835ab, 0x17}, {0xa879c24, 0x1, 0x1})
May 21 03:47:55 scanner rdio-scanner[798]:         /home/chuot/go/src/errors/errors.go:59
[... cut ...]
May 21 03:47:55 scanner rdio-scanner[798]:         /home/chuot/go/src/runtime/os_linux.go:66 +0x61
May 21 03:47:55 scanner rdio-scanner[798]: runtime.futexsleep(0x8ca7130, 0x0, 0xffffffffffffffff)
May 21 03:47:55 scanner rdio-scanner[798]:         /home/chuot/go/src/runtime/sys_linux_386.s:530 +0x1d
May 21 03:47:55 scanner rdio-scanner[798]: runtime.futex()
May 21 03:47:55 scanner rdio-scanner[798]: goroutine 0 [idle]:
May 21 03:47:55 scanner rdio-scanner[798]: PC=0x80aae0d m=0 sigcode=0
May 21 03:47:55 scanner rdio-scanner[798]: SIGQUIT: quit

tdduarte, 2022-05-21:
when i loaded the new ver i just restarted the service... After it crashed I rebooted the instance and its been up and hasnt crashed yet.. so lets see what happens now

tdduarte, 2022-05-21:
upgraded to 6.3.2

tdduarte, 2022-05-22:
Back to 100% still running but maxed out

[image]

tdduarte, 2022-05-24:
Well all of sudden the cpu usage went back to normal

[image]

tdduarte, 2022-05-24:
Great! Thank you!

tdduarte, 2022-05-26:
Still seeing this in 6.3.3

But will upgrade... seems to taking longer to happen now though

[image]

tdduarte, 2022-05-26:
Back up again on .5

But nice not user

[image]

chuot, 2022-05-26:
Unfortunately, this graph doesn't tell if the culprit is Rdio Scanner, it may be due to some other processes running inside the VM. Can you confirm that the rdio-scanner process is really eating up the CPU and if it really is rdio-scanner, try to match its logs to when the VM started running at full throttle.

tdduarte, 2022-05-26:
[image]

tdduarte, 2022-05-26:
I have been looking havent seem to find anything but will keep looking... hard to tell when to look unless it just happens when im watching which hasnt happened yet

josephdambrosio, 2022-05-28:
Can confirm that I'm seeing this issue as well on a VM, Rdio scanner is the only thing that is installed

chuot, 2022-05-31:
Try updating to v6.4.0 to see if the issue still persist.

josephdambrosio, 2022-05-31:
Installed, will advise

tdduarte, 2022-06-01:
Still happening on 6.4

[image]

josephdambrosio, 2022-06-02:
[image]

Same here, pinning up in the 90s

tdduarte, 2022-06-11:
I had to roll back to 6.4.0 the last two updates are just not stable here for whatever reason...

tdduarte, 2022-06-13:
Is it maybe something with using an intel processor?  Do u want access to the instance?  I can email you directly the login info if you are able to provide the IP your coming in from so i can white list it

tdduarte, 2022-06-13:
Cpu has been up for about 24 hours and now its stopped working completely so good time to check it out

chuot, 2022-06-13:
Try v6.4.3

josephdambrosio, 2022-06-18:
Just upgraded will advise

tdduarte, 2022-06-18:
I have been running it for about 3 days how and i think you got it... CPU is staying nice and low and its the longest its been up with out touching it in a while.. Thank you for all your effort on this!

tdduarte, 2022-06-21:
darn... it lasted for so long this time

[image]

chuot, 2022-06-22:
Is this the case with the latest version? A few things have changed about this since v6.4.3.

josephdambrosio, 2022-06-27:
Updated to the newest version and still eventually pinning out at 100%

chuot, 2022-06-27:
@josephdambrosio without giving precise information on your setup, I can't help.

Note that Rdio Scanner can use all the available CPU like 400% on a 4 CPU VM if it has to deal with many API connections at once, which can be the case if you didn't stopped trunk-recorder while installing the latest version of Rdio Scanner. Then when you restart Rdio Scanner, trunk-recorder will send all its pending uploads to Rdio Scanner which it will accept and process right away. Eventually the CPU consumption will go down as the surge is being all processed.

I'm closing this issue because it went too wide with too few details. Reopen a new one with clear details in regards to your issue, what you've tried to identify the cause and please provide logs.

closed 2022-06-27 by chuot