chuot/rdio-scanner #182: Program stops reading in audio files

infusopharm, 2022-05-18:
Running on Windows computer. sdrtrunk working well. For some reason rdio-scan will stop reading in files generated by sdrtrunk. Seems to happen daily. Restart program and all the backed up files get read in with no issues. Not sure if any configuration I may need to address on my end?
Thanks
Steven

infusopharm, 2022-05-18:
Just started experimenting with the program.  I will definately work to provide as much info as I can to determine the issue. Preliminary believe to occur shortly after make changes in the admin page.

chuot, 2022-05-19:
dirwatch seems to be buggy, I'm in the process of rewritting it a little bit. Keep an eye on the upcoming release.

infusopharm, 2022-05-19:
Great.

chuot, 2022-05-20:
try with v6.3.1 to see if it solves your issue

closed 2022-05-26 by chuot

infusopharm, 2022-10-11:
Ran new file. Only runs for a few minutes and then program closes
completely.
Steven

infusopharm, 2022-10-11:
I saw you updated. Will try later today. Much appreciated.