chuot/rdio-scanner #186: Enhancement -  Making the 5 Last Calls Clickable/Replayable

noamlivne, 2022-05-21:
Would it be possible to make the 5 last calls that appear in the Main view clickable and replayable?
That way one can return to those last calls immediately.
I know that there is an option of Search Call, but it contains ALL the calls, unfiltered, and it can take time to find what you just heard.
Thank you.

chuot, 2022-05-21:
Unfortunately, I don't want to make anything clickable on the main screen. The reason for this is that the aim of Rdio Scanner is to mimic real police radio scanners, where the main display is usually not clickable. This is the same reason why I don't show call duration on the main screen, as it breaks the illusion of a real police scanner.

However, I understand the need to have such functionality. That's why I'm going to add it another way. If you press the REPLAY button multiple times within 750ms between each press, you will replay the history according to the number of times you pressed the REPLAY button.

Suppose you only press the REPLAY button once, it will behave as usual. If you press REPLAY twice, it will replay that call on the second line of the history. Press it 3 times, you will replay the call on the third line, and so on.

noamlivne, 2022-05-21:
Thank you very much! It works in the browser (Chrome for Windows and Chrome for Android), but it does not work in the Android app. I guess that is something else?

tdduarte, 2022-05-21:
I think he needs to update the apps. Doesn't work in IOS app either.

chuot, 2022-05-24:
The native app is a different beast and coded with a different language and framework. I am porting this feature. Wait for v1.6.0 which should be released in a few days.

noamlivne, 2022-05-26:
Works fine! Thank you very much. I wonder if the active call within the 5 last calls list can be highlighted when replaying it?
It can be confusing which call is active.
:)

chuot, 2022-05-26:
I'll see what I can do for future releases of the open-source, then I will have to port the change on the native apps.

Since the feature is now added and working, I will close this issue.

closed 2022-05-26 by chuot