chuot/rdio-scanner #187: Directory Watch Crash in the 2 most recent versions

Nokoa, 2022-05-21:
I have updated to the now latest version. I have been constantly getting errors while with a directory watch with SDRTrunk.
This happens every time after a couple minutes of run time with calls uploading. I get this error running the linux version on Ubuntu 20.04 LTS :

```
fatal error: concurrent map writes

goroutine 26 [running]:
runtime.throw({0x8680b1f, 0x15})
	/home/chuot/go/src/runtime/panic.go:992 +0x6a fp=0xadce734 sp=0xadce720 pc=0x807c0ea
runtime.mapassign_faststr(0x86189c0, 0xad505a0, {0xac248a0, 0x5e})
	/home/chuot/go/src/runtime/map_faststr.go:212 +0x36e fp=0xadce778 sp=0xadce734 pc=0x8058b4e
main.(*Dirwatch).Start.func1()
	/home/chuot/Dev/rdio-scanner/server/dirwatch.go:605 +0x32e fp=0xadce7f0 sp=0xadce778 pc=0x85bd21e
runtime.goexit()
	/home/chuot/go/src/runtime/asm_386.s:1326 +0x1 fp=0xadce7f4 sp=0xadce7f0 pc=0x80a9c71
created by main.(*Dirwatch).Start
	/home/chuot/Dev/rdio-scanner/server/dirwatch.go:554 +0x1b4

goroutine 1 [IO wait, 8 minutes]:
internal/poll.runtime_pollWait(0xe7c49f54, 0x72)
	/home/chuot/go/src/runtime/netpoll.go:302 +0x52
internal/poll.(*pollDesc).wait(0xae20014, 0x72, 0x0)
	/home/chuot/go/src/internal/poll/fd_poll_runtime.go:83 +0x37
internal/poll.(*pollDesc).waitRead(...)
	/home/chuot/go/src/internal/poll/fd_poll_runtime.go:88
internal/poll.(*FD).Accept(0xae20000)
	/home/chuot/go/src/internal/poll/fd_unix.go:614 +0x223
net.(*netFD).accept(0xae20000)
	/home/chuot/go/src/net/fd_unix.go:172 +0x27
net.(*TCPListener).accept(0xae10040)
	/home/chuot/go/src/net/tcpsock_posix.go:139 +0x27
net.(*TCPListener).Accept(0xae10040)
	/home/chuot/go/src/net/tcpsock.go:288 +0x37
net/http.(*Server).Serve(0xae18090, {0x88eb0d8, 0xae10040})
	/home/chuot/go/src/net/http/server.go:3039 +0x32d
net/http.(*Server).ListenAndServe(0xae18090)
	/home/chuot/go/src/net/http/server.go:2968 +0xa4
main.main()
	/home/chuot/Dev/rdio-scanner/server/main.go:244 +0xe1e

goroutine 18 [select, 8 minutes]:
database/sql.(*DB).connectionOpener(0xaca4990, {0x88eb378, 0xacd2330})
	/home/chuot/go/src/database/sql/sql.go:1226 +0xab
created by database/sql.OpenDB
	/home/chuot/go/src/database/sql/sql.go:794 +0x184

goroutine 19 [select]:
database/sql.(*DB).connectionCleaner(0xaca4990, 0xdf8475800)
	/home/chuot/go/src/database/sql/sql.go:1069 +0xdf
created by database/sql.(*DB).startCleanerLocked
	/home/chuot/go/src/database/sql/sql.go:1056 +0x17e

goroutine 20 [select, 8 minutes]:
main.(*Admin).Start.func1()
	/home/chuot/Dev/rdio-scanner/server/admin.go:582 +0xae
created by main.(*Admin).Start
	/home/chuot/Dev/rdio-scanner/server/admin.go:580 +0x6e

goroutine 21 [select, 8 minutes]:
main.(*Scheduler).Start.func1()
	/home/chuot/Dev/rdio-scanner/server/scheduler.go:89 +0x93
created by main.(*Scheduler).Start
	/home/chuot/Dev/rdio-scanner/server/scheduler.go:87 +0xab

goroutine 22 [chan receive, 8 minutes]:
main.(*Controller).Start.func1()
	/home/chuot/Dev/rdio-scanner/server/controller.go:606 +0x96
created by main.(*Controller).Start
	/home/chuot/Dev/rdio-scanner/server/controller.go:603 +0x325

goroutine 23 [syscall]:
syscall.Syscall6(0x11c, 0x1, 0x395260, 0xb20fcec, 0x1000004, 0x0, 0x0)
	/home/chuot/go/src/syscall/asm_linux_386.s:44 +0x5
os.(*Process).blockUntilWaitable(0xb2d2540)
	/home/chuot/go/src/os/wait_waitid.go:32 +0x73
os.(*Process).wait(0xb2d2540)
	/home/chuot/go/src/os/exec_unix.go:22 +0x2e
os.(*Process).Wait(...)
	/home/chuot/go/src/os/exec.go:132
os/exec.(*Cmd).Wait(0xad02b00)
	/home/chuot/go/src/os/exec/exec.go:507 +0x53
os/exec.(*Cmd).Run(0xad02b00)
	/home/chuot/go/src/os/exec/exec.go:341 +0x49
main.(*Controller).ConvertAudio(0xacda840, 0xb214140)
	/home/chuot/Dev/rdio-scanner/server/controller.go:128 +0xa75
main.(*Controller).IngestCall(0xacda840, 0xb214140)
	/home/chuot/Dev/rdio-scanner/server/controller.go:349 +0x1154
main.(*Controller).Start.func2()
	/home/chuot/Dev/rdio-scanner/server/controller.go:613 +0x4b
created by main.(*Controller).Start
	/home/chuot/Dev/rdio-scanner/server/controller.go:610 +0x36a

goroutine 24 [select, 8 minutes]:
main.(*Controller).Start.func3()
	/home/chuot/Dev/rdio-scanner/server/controller.go:634 +0xdb
created by main.(*Controller).Start
	/home/chuot/Dev/rdio-scanner/server/controller.go:617 +0x3af

goroutine 25 [select]:
github.com/fsnotify/fsnotify.(*Watcher).readEvents(0xacd2de0)
	/home/chuot/go/pkg/mod/github.com/fsnotify/fsnotify@v1.5.4/inotify.go:298 +0x8ff
created by github.com/fsnotify/fsnotify.NewWatcher
	/home/chuot/go/pkg/mod/github.com/fsnotify/fsnotify@v1.5.4/inotify.go:60 +0x1e7

goroutine 49 [chan send]:
main.(*Dirwatch).ingestSdrTrunk(0xacaafc0, {0xb1ce0e0, 0x61})
	/home/chuot/Dev/rdio-scanner/server/dirwatch.go:243 +0x428
main.(*Dirwatch).Ingest(0xacaafc0, {0xb1ce0e0, 0x61})
	/home/chuot/Dev/rdio-scanner/server/dirwatch.go:140 +0xf1
main.(*Dirwatch).Start.func2.1({0xb3fc400, 0x40}, {0x88eb538, 0xb042260}, {0x0, 0x0})
	/home/chuot/Dev/rdio-scanner/server/dirwatch.go:629 +0xcb
io/fs.walkDir({0x88ea1a8, 0xadbc930}, {0xb3fc400, 0x40}, {0x88eb538, 0xb042260}, 0xad68fe4)
	/home/chuot/go/src/io/fs/walk.go:66 +0x50
io/fs.walkDir({0x88ea1a8, 0xadbc930}, {0x8673d0e, 0x1}, {0x88eb498, 0xadbc938}, 0xad68fe4)
	/home/chuot/go/src/io/fs/walk.go:85 +0x281
io/fs.WalkDir({0x88ea1a8, 0xadbc930}, {0x8673d0e, 0x1}, 0xae0a7e4)
	/home/chuot/go/src/io/fs/walk.go:112 +0x10e
main.(*Dirwatch).Start.func2()
	/home/chuot/Dev/rdio-scanner/server/dirwatch.go:621 +0xa1
created by main.(*Dirwatch).Start
	/home/chuot/Dev/rdio-scanner/server/dirwatch.go:618 +0x21a

goroutine 5 [syscall, 8 minutes]:
os/signal.signal_recv()
	/home/chuot/go/src/runtime/sigqueue.go:151 +0x3b
os/signal.loop()
	/home/chuot/go/src/os/signal/signal_unix.go:23 +0x1a
created by os/signal.Notify.func1.1
	/home/chuot/go/src/os/signal/signal.go:151 +0x27

goroutine 10337 [IO wait]:
internal/poll.runtime_pollWait(0xe7c49eb8, 0x72)
	/home/chuot/go/src/runtime/netpoll.go:302 +0x52
internal/poll.(*pollDesc).wait(0xac598d4, 0x72, 0x1)
	/home/chuot/go/src/internal/poll/fd_poll_runtime.go:83 +0x37
internal/poll.(*pollDesc).waitRead(...)
	/home/chuot/go/src/internal/poll/fd_poll_runtime.go:88
internal/poll.(*FD).Read(0xac598c0, {0xb2ca600, 0x200, 0x200})
	/home/chuot/go/src/internal/poll/fd_unix.go:167 +0x1e0
os.(*File).read(...)
	/home/chuot/go/src/os/file_posix.go:31
os.(*File).Read(0xada03a8, {0xb2ca600, 0x200, 0x200})
	/home/chuot/go/src/os/file.go:119 +0x62
bytes.(*Buffer).ReadFrom(0xb1bcf30, {0x88e9a00, 0xada03a8})
	/home/chuot/go/src/bytes/buffer.go:204 +0x95
io.copyBuffer({0x88e935c, 0xb1bcf30}, {0x88e9a00, 0xada03a8}, {0x0, 0x0, 0x0})
	/home/chuot/go/src/io/io.go:412 +0x12b
io.Copy(...)
	/home/chuot/go/src/io/io.go:385
os/exec.(*Cmd).writerDescriptor.func1()
	/home/chuot/go/src/os/exec/exec.go:311 +0x54
os/exec.(*Cmd).Start.func1(0xae11810)
	/home/chuot/go/src/os/exec/exec.go:441 +0x24
created by os/exec.(*Cmd).Start
	/home/chuot/go/src/os/exec/exec.go:440 +0x6a8

goroutine 10320 [IO wait]:
internal/poll.runtime_pollWait(0xe7c49c48, 0x72)
	/home/chuot/go/src/runtime/netpoll.go:302 +0x52
internal/poll.(*pollDesc).wait(0xac59854, 0x72, 0x1)
	/home/chuot/go/src/internal/poll/fd_poll_runtime.go:83 +0x37
internal/poll.(*pollDesc).waitRead(...)
	/home/chuot/go/src/internal/poll/fd_poll_runtime.go:88
internal/poll.(*FD).Read(0xac59840, {0xb2ca400, 0x200, 0x200})
	/home/chuot/go/src/internal/poll/fd_unix.go:167 +0x1e0
os.(*File).read(...)
	/home/chuot/go/src/os/file_posix.go:31
os.(*File).Read(0xada0380, {0xb2ca400, 0x200, 0x200})
	/home/chuot/go/src/os/file.go:119 +0x62
bytes.(*Buffer).ReadFrom(0xb1bcf18, {0x88e9a00, 0xada0380})
	/home/chuot/go/src/bytes/buffer.go:204 +0x95
io.copyBuffer({0x88e935c, 0xb1bcf18}, {0x88e9a00, 0xada0380}, {0x0, 0x0, 0x0})
	/home/chuot/go/src/io/io.go:412 +0x12b
io.Copy(...)
	/home/chuot/go/src/io/io.go:385
os/exec.(*Cmd).writerDescriptor.func1()
	/home/chuot/go/src/os/exec/exec.go:311 +0x54
os/exec.(*Cmd).Start.func1(0xae11800)
	/home/chuot/go/src/os/exec/exec.go:441 +0x24
created by os/exec.(*Cmd).Start
	/home/chuot/go/src/os/exec/exec.go:440 +0x6a8

goroutine 10338 [chan send]:
main.(*Dirwatch).ingestSdrTrunk(0xacaafc0, {0xb1e8540, 0x61})
	/home/chuot/Dev/rdio-scanner/server/dirwatch.go:243 +0x428
main.(*Dirwatch).Ingest(0xacaafc0, {0xb1e8540, 0x61})
	/home/chuot/Dev/rdio-scanner/server/dirwatch.go:140 +0xf1
main.(*Dirwatch).Start.func1.2.1()
	/home/chuot/Dev/rdio-scanner/server/dirwatch.go:564 +0x7f
created by time.goFunc
	/home/chuot/go/src/time/sleep.go:176 +0x2f

```

noamlivne, 2022-05-21:
Also in Windows the program's terminal window disappears after a few minutes. I did not catch any error. Happens with the latest 2 versions (3.1 and 3.2). Does not happen in 3.0.

infusopharm, 2022-05-21:
Having the same issue with both 6.3.2 and 6.3.1 in Windows. Window closes and program ends. 
 Currently back to running 6.2.5.

Bobabadabouskie, 2022-05-24:
Similar error/crash on Debian 11.

chuot, 2022-05-24:
Please run it in a cmd window, then provide me with the crash log.

chuot, 2022-05-25:
Open a CMD window, change directory to where Rdio Scanner is installed, then simply run it at the command prompt by typing `rdio-scanner`. It should then log on the standard output. When it will crash, copy all the outputs and post them here. Without the crash log, I can't help.

Nokoa, 2022-05-25:
Using your latest 6.3.4 now. Will report back.

Nokoa, 2022-05-25:
After running for 6 minutes or so minutes, uploading calls during the entire time, successfully. Rdio Scanner crashes with the following error:

```
fatal error: concurrent map writes

goroutine 27 [running]:
runtime.throw({0x8681c5e, 0x15})
	/home/chuot/go/src/runtime/panic.go:992 +0x6a fp=0xae5270c sp=0xae526f8 pc=0x807c0ea
runtime.mapassign_faststr(0x8619a20, 0xaeb4000, {0xac24a80, 0x5e})
	/home/chuot/go/src/runtime/map_faststr.go:212 +0x36e fp=0xae52750 sp=0xae5270c pc=0x8058b4e
main.(*Dirwatch).Start.func1()
	/home/chuot/Dev/rdio-scanner/server/dirwatch.go:614 +0x3a2 fp=0xae527f0 sp=0xae52750 pc=0x85bc792
runtime.goexit()
	/home/chuot/go/src/runtime/asm_386.s:1326 +0x1 fp=0xae527f4 sp=0xae527f0 pc=0x80a9c71
created by main.(*Dirwatch).Start
	/home/chuot/Dev/rdio-scanner/server/dirwatch.go:549 +0x1b4

goroutine 1 [IO wait, 4 minutes]:
internal/poll.runtime_pollWait(0xe7c89f54, 0x72)
	/home/chuot/go/src/runtime/netpoll.go:302 +0x52
internal/poll.(*pollDesc).wait(0xacbd554, 0x72, 0x0)
	/home/chuot/go/src/internal/poll/fd_poll_runtime.go:83 +0x37
internal/poll.(*pollDesc).waitRead(...)
	/home/chuot/go/src/internal/poll/fd_poll_runtime.go:88
internal/poll.(*FD).Accept(0xacbd540)
	/home/chuot/go/src/internal/poll/fd_unix.go:614 +0x223
net.(*netFD).accept(0xacbd540)
	/home/chuot/go/src/net/fd_unix.go:172 +0x27
net.(*TCPListener).accept(0xad35a10)
	/home/chuot/go/src/net/tcpsock_posix.go:139 +0x27
net.(*TCPListener).Accept(0xad35a10)
	/home/chuot/go/src/net/tcpsock.go:288 +0x37
net/http.(*Server).Serve(0xaca4cf0, {0x88ec4c0, 0xad35a10})
	/home/chuot/go/src/net/http/server.go:3039 +0x32d
net/http.(*Server).ListenAndServe(0xaca4cf0)
	/home/chuot/go/src/net/http/server.go:2968 +0xa4
main.main()
	/home/chuot/Dev/rdio-scanner/server/main.go:245 +0xe1d

goroutine 19 [select, 6 minutes]:
database/sql.(*DB).connectionOpener(0xaca4bd0, {0x88ec760, 0xacd4360})
	/home/chuot/go/src/database/sql/sql.go:1226 +0xab
created by database/sql.OpenDB
	/home/chuot/go/src/database/sql/sql.go:794 +0x184

goroutine 20 [select]:
database/sql.(*DB).connectionCleaner(0xaca4bd0, 0xdf8475800)
	/home/chuot/go/src/database/sql/sql.go:1069 +0xdf
created by database/sql.(*DB).startCleanerLocked
	/home/chuot/go/src/database/sql/sql.go:1056 +0x17e

goroutine 21 [select, 6 minutes]:
main.(*Admin).Start.func1()
	/home/chuot/Dev/rdio-scanner/server/admin.go:576 +0xae
created by main.(*Admin).Start
	/home/chuot/Dev/rdio-scanner/server/admin.go:574 +0x6e

goroutine 22 [select, 6 minutes]:
main.(*Scheduler).Start.func1()
	/home/chuot/Dev/rdio-scanner/server/scheduler.go:85 +0x93
created by main.(*Scheduler).Start
	/home/chuot/Dev/rdio-scanner/server/scheduler.go:83 +0xab

goroutine 23 [chan receive, 6 minutes]:
main.(*Controller).Start.func1()
	/home/chuot/Dev/rdio-scanner/server/controller.go:521 +0x96
created by main.(*Controller).Start
	/home/chuot/Dev/rdio-scanner/server/controller.go:518 +0x2a8

goroutine 24 [chan receive]:
main.(*Controller).Start.func2()
	/home/chuot/Dev/rdio-scanner/server/controller.go:527 +0x37
created by main.(*Controller).Start
	/home/chuot/Dev/rdio-scanner/server/controller.go:525 +0x2ed

goroutine 25 [select, 3 minutes]:
main.(*Controller).Start.func3()
	/home/chuot/Dev/rdio-scanner/server/controller.go:549 +0xdb
created by main.(*Controller).Start
	/home/chuot/Dev/rdio-scanner/server/controller.go:532 +0x332

goroutine 26 [select]:
github.com/fsnotify/fsnotify.(*Watcher).readEvents(0xacd4e10)
	/home/chuot/go/pkg/mod/github.com/fsnotify/fsnotify@v1.5.4/inotify.go:298 +0x8ff
created by github.com/fsnotify/fsnotify.NewWatcher
	/home/chuot/go/pkg/mod/github.com/fsnotify/fsnotify@v1.5.4/inotify.go:60 +0x1e7

goroutine 34 [syscall, 6 minutes]:
os/signal.signal_recv()
	/home/chuot/go/src/runtime/sigqueue.go:151 +0x3b
os/signal.loop()
	/home/chuot/go/src/os/signal/signal_unix.go:23 +0x1a
created by os/signal.Notify.func1.1
	/home/chuot/go/src/os/signal/signal.go:151 +0x27

goroutine 2902 [runnable]:
main.NewCall(...)
	/home/chuot/Dev/rdio-scanner/server/call.go:51
main.(*Dirwatch).ingestSdrTrunk(0xacaaee0, {0xb261500, 0x5f})
	/home/chuot/Dev/rdio-scanner/server/dirwatch.go:223 +0x173
main.(*Dirwatch).Ingest(0xacaaee0, {0xb261500, 0x5f})
	/home/chuot/Dev/rdio-scanner/server/dirwatch.go:139 +0xf1
main.(*Dirwatch).Start.func1.2.1()
	/home/chuot/Dev/rdio-scanner/server/dirwatch.go:561 +0x7f
created by time.goFunc
	/home/chuot/go/src/time/sleep.go:176 +0x2f

```

chuot, 2022-05-25:
It doesn't look like v6.3.4. Please verify the version number with this command: `rdio-scanner -version`.

chuot, 2022-05-25:
Disregard the last comment. I just saw the error.

chuot, 2022-05-25:
Let me know if it still running after 15 minutes

infusopharm, 2022-05-25:
Still could not find error event after window closed. Currently running 6.3.5 for 2 hrs with no issues. We'll done.

Nokoa, 2022-05-26:
Running the latest so far for 3 hours with no issues.

closed 2022-05-26 by chuot