chuot/rdio-scanner #189: ffmpeg not working on Debian 10.12 with 6.3.3

kb2ear, 2022-05-24:
```
May 24 05:59:43 rdio-instance1 Rdio Scanner 10338[5612]: ffmpeg version 4.1.9-0+deb10u1 Copyright (c) 2000-2022 the FFmpeg developers
May 24 05:59:43 rdio-instance1 Rdio Scanner 10338[5612]:   built with gcc 8 (Debian 8.3.0-6)
May 24 05:59:43 rdio-instance1 Rdio Scanner 10338[5612]:   configuration: --prefix=/usr --extra-version=0+deb10u1 --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --arch=amd64 --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared
May 24 05:59:43 rdio-instance1 Rdio Scanner 10338[5612]:   libavutil      56. 22.100 / 56. 22.100
May 24 05:59:43 rdio-instance1 Rdio Scanner 10338[5612]:   libavcodec     58. 35.100 / 58. 35.100
May 24 05:59:43 rdio-instance1 Rdio Scanner 10338[5612]:   libavformat    58. 20.100 / 58. 20.100
May 24 05:59:43 rdio-instance1 Rdio Scanner 10338[5612]:   libavdevice    58.  5.100 / 58.  5.100
May 24 05:59:43 rdio-instance1 Rdio Scanner 10338[5612]:   libavfilter     7. 40.101 /  7. 40.101
May 24 05:59:43 rdio-instance1 Rdio Scanner 10338[5612]:   libavresample   4.  0.  0 /  4.  0.  0
May 24 05:59:43 rdio-instance1 Rdio Scanner 10338[5612]:   libswscale      5.  3.100 /  5.  3.100
May 24 05:59:43 rdio-instance1 Rdio Scanner 10338[5612]:   libswresample   3.  3.100 /  3.  3.100
May 24 05:59:43 rdio-instance1 Rdio Scanner 10338[5612]:   libpostproc    55.  3.100 / 55.  3.100
May 24 05:59:43 rdio-instance1 Rdio Scanner 10338[5612]: Input #0, mov,mp4,m4a,3gp,3g2,mj2, from 'pipe:':
May 24 05:59:43 rdio-instance1 Rdio Scanner 10338[5612]:   Metadata:
May 24 05:59:43 rdio-instance1 Rdio Scanner 10338[5612]:     major_brand     : M4A
May 24 05:59:43 rdio-instance1 Rdio Scanner 10338[5612]:     minor_version   : 512
May 24 05:59:43 rdio-instance1 Rdio Scanner 10338[5612]:     compatible_brands: isomiso2iso6
May 24 05:59:43 rdio-instance1 Rdio Scanner 10338[5612]:     encoder         : Lavf57.83.100
May 24 05:59:43 rdio-instance1 Rdio Scanner 10338[5612]:   Duration: 00:00:13.81, start: 0.000000, bitrate: N/A
May 24 05:59:43 rdio-instance1 Rdio Scanner 10338[5612]:     Stream #0:0(und): Audio: aac (LC) (mp4a / 0x6134706D), 8000 Hz, mono, fltp, 38 kb/s (default)
May 24 05:59:43 rdio-instance1 Rdio Scanner 10338[5612]:     Metadata:
May 24 05:59:43 rdio-instance1 Rdio Scanner 10338[5612]:       handler_name    : SoundHandler
May 24 05:59:43 rdio-instance1 Rdio Scanner 10338[5612]: Stream mapping:
May 24 05:59:43 rdio-instance1 Rdio Scanner 10338[5612]:   Stream #0:0 -> #0:0 (aac (native) -> aac (native))
May 24 05:59:43 rdio-instance1 Rdio Scanner 10338[5612]: [Parsed_apad_0 @ 0x56469d3bab00] Option 'whole_dur' not found
May 24 05:59:43 rdio-instance1 Rdio Scanner 10338[5612]: [AVFilterGraph @ 0x56469d325840] Error initializing filter 'apad' with args 'whole_dur=3s'
May 24 05:59:43 rdio-instance1 Rdio Scanner 10338[5612]: Error reinitializing filters!
May 24 05:59:43 rdio-instance1 Rdio Scanner 10338[5612]: Failed to inject frame into filter network: Option not found
May 24 05:59:43 rdio-instance1 Rdio Scanner 10338[5612]: Error while processing the decoded data for stream #0:0
May 24 05:59:43 rdio-instance1 Rdio Scanner 10338[5612]: Conversion failed!
May 24 05:59:43 rdio-instance1 Rdio Scanner 10338[5612]: 2022/05/24 05:59:43 newcall: system=10338 talkgroup=119 file=119-1653411549_4604500001.m4a success
```
Debian version: 
```
Linux rdio-instance1 4.19.0-20-cloud-amd64 #1 SMP Debian 4.19.235-1 (2022-03-17) x86_64 GNU/Linux
```
ffmpeg version:
```
ffmpeg version 4.1.9-0+deb10u1 Copyright (c) 2000-2022 the FFmpeg developers
  built with gcc 8 (Debian 8.3.0-6)
  configuration: --prefix=/usr --extra-version=0+deb10u1 --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --arch=amd64 --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared
  libavutil      56. 22.100 / 56. 22.100
  libavcodec     58. 35.100 / 58. 35.100
  libavformat    58. 20.100 / 58. 20.100
  libavdevice    58.  5.100 / 58.  5.100
  libavfilter     7. 40.101 /  7. 40.101
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  3.100 /  5.  3.100
  libswresample   3.  3.100 /  3.  3.100
  libpostproc    55.  3.100 / 55.  3.100
```

chuot, 2022-05-24:
Try updating ffmpeg to v5.0.1 or disable audio conversion for now

chuot, 2022-05-24:
ffmpeg 4.3.3 on debian 11 works fyi. The problem is the whole_dur parameter for the apad filter that don't yet exists with ffmpeg v4.1 on Debian 10.

chuot, 2022-05-25:
Starting with v6.3.4, I am now testing the ffmpeg version and adapting the arguments accordingly. ffmpeg version lower than 4.3 will not use audio normalization.

closed 2022-05-25 by kb2ear