chuot/rdio-scanner #190: Settings in options not being read at startup

kb2ear, 2022-05-25:
discovered that anything you set in the options like: Disable Audio Conversion, Show Listeners Count to be specific, are not being held across restarts of the software going back to at least 6.3.2. I can set them, they work until I restart the software, then they no longer show as turned on and are not working.  If I do a export config the json shows them as true.

kb2ear, 2022-05-25:
export before restart:
  "options": {
    "afsSystems": "",
    "autoPopulate": true,
    "dimmerDelay": 5000,
    "disableAudioConversion": true,
    "disableDuplicateDetection": false,
    "duplicateDetectionTimeFrame": 500,
    "keypadBeeps": "uniden",
    "maxClients": 200,
    "pruneDays": 7,
    "searchPatchedTalkgroups": false,
    "showListenersCount": true,
    "sortTalkgroups": false,
    "tagsToggle": false
  },
```
export after restart:
```
  "options": {
    "afsSystems": "",
    "autoPopulate": true,
    "dimmerDelay": 5000,
    "disableAudioConversion": false,
    "disableDuplicateDetection": false,
    "duplicateDetectionTimeFrame": 500,
    "keypadBeeps": "uniden",
    "maxClients": 200,
    "pruneDays": 7,
    "searchPatchedTalkgroups": false,
    "showListenersCount": false,
    "sortTalkgroups": false,
    "tagsToggle": false
  },
```
in SQL after restart:
```
{
   "pruneDays":7,
   "afsSystems":"",
   "maxClients":200,
   "tagsToggle":false,
   "dimmerDelay":5000,
   "keypadBeeps":"uniden",
   "autoPopulate":true,
   "sortTalkgroups":false,
   "showListenersCount":true,
   "disableAudioConversion":true,
   "searchPatchedTalkgroups":false,
   "disableDuplicateDetection":false,
   "duplicateDetectionTimeFrame":500
}
```

kb2ear, 2022-05-25:
same issue on mariadb

chuot, 2022-05-25:
it's actually the same code. Please let me know if v6.3.5 fixed it

kb2ear, 2022-05-25:
I seems to be fixed in 6.3.5 thanks

closed 2022-05-25 by kb2ear