chuot/rdio-scanner #191: 6.3.3 and Rdio App - No Audio

JayManCanada6014, 2022-05-26:
I have Rdio App running 24/7 on a tablet on my desk, since the 6.3.x from the last days troubleshooting those issues now every so often the audio stops on my android tablet.   The display keeps going, I see calls playing but no audio.  
I did update the app today but the same issue.

If I push the buttons on the screen there is audio, just no scanner audio..   Website works fine,  not sure if anyone else is having the issue.

JayManCanada6014, 2022-05-26:
Need to look at something first.

JayManCanada6014, 2022-05-26:
So this was happening on 6.3.3 and it's happening on 6.3.5.   
Samsung Galaxy Tab E 8.0  with Android 7.1.1 (last version available)
Fully Updated

chuot, 2022-05-26:
Please try to exclude your VPS provider from the equation. I noticed your connection was laggy 2 days ago when I monitored it. The issue you describe can occur when the connection between client and server has been lost and the keepalive on the websocket has not yet occurred. During this time it will not show nolink even if the connection is lost somewhere in the middle.

Do the server logs show anything interesting when this happens? Again, without logs it's very hard to troubleshoot.

JayManCanada6014, 2022-05-26:
Logs show nothing, it's not server side its just keeps going like normal, the app just stops playing audio, the rest of the app works as normal just no audio plays.

The laggy you were seeing was cloudflare because I thought I was being DDOS'd when I had those last issues so I had the proxy turned on.  Its all off now.

noamlivne, 2022-05-30:
Perhaps one of the issues here is similar to my Android issue where the audio disappears but everything else seems to work.
Could you give me a clue which line would you want me to enter in ADB LOGCAT? Or simply give you .txt file with any mention of rdioScanner?

JayManCanada6014, 2022-05-30:
> Perhaps one of the issues here is similar to my Android issue where the audio disappears but everything else seems to work. Could you give me a clue which line would you want me to enter in ADB LOGCAT? Or simply give you .txt file with any mention of rdioScanner?

That is exactly what happens for me.   Just the audio stops, the calls still show on the display etc..

chuot, 2022-05-30:
Oh, I missed the fact that the display continue but not the audio, so it must be something with the audio library I'm using. It also looks related to android only. I no longer daily drive an android device, but I have one for testing so I will set a test bench just for that issue and will let you know.

JayManCanada6014, 2022-05-30:
> Well, the android device that was running for a long time on my desk had the issue. I don't for how long it went silent and I found nothing in the catlog. So I've restarted a debug version on the same device and I will let it run until I catch the error. It may be due to a memory leak in the underlying audio library.
> 
> I will also speed up the new audio library integration that I started to do a couple of weeks ago. The current audio library has been deprecated, so no luck for a quick fix.

It only started for me when we did the work from 6.3.1 to 6.3.6, before that there were no issues.

noamlivne, 2022-05-31:
For me this issue is much older. Maybe from the beginning of my usage of Android (from version 6). But any improvement will be greatly appreciated.

chuot, 2022-05-31:
I've been able to reproduce after some time and there's some odd logcat errors which relate to the mediaplayer. Still looking to fix the issue which seems to affects android only.

noamlivne, 2022-07-03:
After a few days with the new Android version (and 6.5.1 in Windows) , I think this specific issue of audio-stopping-and-cannot-be-revived is solved!
Thank you very much!

There are, unfortunately, two "new" issues with the audio in my Android. (Perhaps you would like me to open a New Issue):
1. The audio does stop rarely, but this time, pressing Skip Next - helps - so that is not a big issue at all.
2. Sometimes, the audio is clipped at the last second of each over. In SDRTrunk it sounds ok. 
So I guess it is coming from the the Android app. Swiping the app from the memory and starting it again - solves it.

I hope others can chime in, if they encounter these.

chuot, 2022-07-04:
@noamlivne Please send me your email address linked to your google account, I will add you to the beta channel.

noamlivne, 2022-07-05:
1. I did not encounter "The audio does stop rarely, but this time, pressing Skip Next" (even in 1.7 I tested a day after I wrote this issue) - so it is rare.

2. I did encounter the last-second clipping quite a lot in 1.7 and 1.8 in XiaoMi Redmi 9T Android version 11 (MIUI Global 12.5.11). Now I think that swiping the app from memory does not help. 
The clipping happens "live" and in the archive of the Android app. 
There is no clipping in Live SDRTrunk stream, or in the Recordings from SDRTrunk, or in the Browser version of RDIO-Scanner.

I did not encounter the last second clipping in 1.7 in XiaoMi Redmi 4 Android version 6.01 (MIUI Global  10.2).
I could not test 1.8 on that old device because when the first call enters - it crashes the RDIO-Scanner app. I sent you the Crash Report from that device.

noamlivne, 2022-07-06:
I tried the new beta version, and unfortunately the last second clipping still appears. (Tested on the modern Xiaomi, the older Xiaomi still crashes).

noamlivne, 2022-07-06:
Thank you very much!

chuot, 2022-07-11:
The latest build of native v1.8, which is already on the beta channel, will be released on the main channel in a couple of hours.

closed 2022-07-11 by chuot