chuot/rdio-scanner #195: rdio-scanner not shipping calls to downstream

infecticide, 2022-05-28:
Hi Chrystian it's me again, you are likely tired of seeing my name in the issues list!

rdio-scanner server: 6.3.6
DB Type: mariadb
OS: Ubuntu 21.04

A friend of mine has setup a rdio-scanner instance, provided me with an API key and URL.
I have set this up as a "Downstream" and selected the TGs I want to ship out to them.

At no time does my rdio-scanner server seem to reach out to send the calls when they are processed.  I have no errors, warnings or informational logs about calls being sent to the downstream. I have also confirmed via a packet trace on the network that my server never attempts to send the calls downstream.

I have verified the API key, the URL, the TGs and that the Downstream is not disabled.

My friend is sending calls to me successfully.

infecticide, 2022-05-29:
Switching to SQLite backend solves this issue, I would like to continue using MySQL however.

Dewey3, 2022-05-30:
Just a thought... something that tripped me up the first time I used the downstream feature.  I was actually disabling the downstream because I though turning the first slider to on (to the right) was enabling the downstream, not disabling it (EGG ON FACE!).

chuot, 2022-05-30:
It looks like your config is done so it will only downstream talkgroups 5490 and 6175 from system 1. Every other calls that doesn't match either one of the two talkgroups wont be downstreamed at all. If your intention is to downstream everything from system 1, then go back to your downstream configuration and click on `choose systems`. From there, make sure the sys/tg selection is set to what you really want to downstream to your friend. Don't forget to click on `save` after closing the popup window to persist the new configuration.

Note that when downstream part of your whole setup, the downstream definition section will tell: This downstream allows access to **some** systems and talkgroups, as opposed to **all**.

infecticide, 2022-05-30:
That is correct that we only want to ship those 2 TGs.  It doesn't seem to be happening though.

There are calls on those TGs from 07:26 this morning and my logs do not show that it sent those calls downstream.

I've confirmed with the recipient, those calls do not appear downstream.

infecticide, 2022-05-30:
I thought it might be helpful to post my rdio-scanner.service file as well:

[Unit]
Description=The perfect software-defined radio companion
ConditionFileIsExecutable=/home/infecticide/rdio-scanner

[Service]
StartLimitInterval=5
StartLimitBurst=10
ExecStart=/home/infecticide/rdio-scanner "-service" "run" -db_host MARIADBSERVERNAME -db_name rdio_scanner -db_port 3306 -db_type mariadb -db_user rdio_scanner -db_pass PASSWORDREMOVED
User=infecticide
Restart=always
RestartSec=120

[Install]
WantedBy=multi-user.target

chuot, 2022-05-30:
Still trying to figure out the issue.

You should use the -config_save option to save all the database properties instead of passing them on the ExecStart line. Look in the PDF for some examples.

infecticide, 2022-05-30:
I've updated my systemctl unit to use -config instead.

infecticide, 2022-06-27:
I have been running 6.4.5 for a couple days now and my calls are still not shipping downstream.

infecticide, 2022-07-05:
6.5.1 installed, there are no log entries indicating it has even attempted the ship the calls.  I can see a log entry for the dirwatch file retrieval.

I've included some screenshots and my config.
[image]
[image]
[rdio-scanner.txt](https://github.com/chuot/rdio-scanner/files/9049257/rdio-scanner.txt)

infecticide, 2022-07-05:
6.5.1 installed, but calls still not shipping.

chuot, 2022-07-06:
Here's what I tried this morning:

- Installed a fresh Ubuntu Desktop 21.10 VM (host is Fedora WorkStation 26)
- Installed the latest 21.10 updates
- Installed ffmpeg + mariadb-server
- Created the mariadb database for Rdio Scanner
- Installed Rdio Scanner v6.5.1
- ./rdio-scanner -db_type=mariadb -db_name=rdioscanner -db_user=rdioscanner -db_pass=rdioscanner -config_save
- ./rdio-scanner
- Loaded your rdio-scanner.txt configuration through the admin dashboard
- Tweaked the dirwatch folder to match my setup, changed notthing else
- Took a MP3 file directly from a SDRTrunk recordings and tweaked its MP3 tags to match your system 1 talkgroup 5490.
- Copied that tweaked MP3 to the folder monitored by diwatch
- ... and it worked as you can see from the logs below

```bash
chuot@ubuntu2110:~/Rdio Scanner$ uname -a
Linux ubuntu2110 5.13.0-52-generic #59-Ubuntu SMP Wed Jun 15 20:17:13 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
chuot@ubuntu2110:~/Rdio Scanner$ sudo mysql
Welcome to the MariaDB monitor.  Commands end with ; or \g.
Your MariaDB connection id is 31
Server version: 10.5.15-MariaDB-0ubuntu0.21.10.1 Ubuntu 21.10

Copyright (c) 2000, 2018, Oracle, MariaDB Corporation Ab and others.

Type 'help;' or '\h' for help. Type '\c' to clear the current input statement.

MariaDB [(none)]> create database rdioscanner;
Query OK, 1 row affected (0.001 sec)

MariaDB [(none)]> grant all privileges on rdioscanner.* to 'rdioscanner'@'localhost' identified by 'rdioscanner';
Query OK, 0 rows affected (0.007 sec)

MariaDB [(none)]> ^DBye
chuot@ubuntu2110:~/Rdio Scanner$ ls
rdio-scanner-linux-amd64-v6.5.1.zip
chuot@ubuntu2110:~/Rdio Scanner$ unzip -o rdio-scanner-linux-amd64-v6.5.1.zip 
Archive:  rdio-scanner-linux-amd64-v6.5.1.zip
  inflating: rdio-scanner            
  inflating: rdio-scanner.pdf        
chuot@ubuntu2110:~/Rdio Scanner$ ./rdio-scanner -db_type=mariadb -db_name=rdioscanner -db_user=rdioscanner -db_pass=rdioscanner -config_save
rdio-scanner.ini file created
chuot@ubuntu2110:~/Rdio Scanner$ cat rdio-scanner.ini
db_host = localhost
db_name = rdioscanner
db_pass = rdioscanner
db_port = 3306
db_type = mariadb
db_user = rdioscanner
listen = :3000
chuot@ubuntu2110:~/Rdio Scanner$ ./rdio-scanner 

Rdio Scanner v6.5.1
----------------------------------
2022/07/06 09:51:22 server started
2022/07/06 09:51:22 base folder is /home/chuot/Rdio Scanner
2022/07/06 09:51:22 main interface at http://ubuntu2110:3000
2022/07/06 09:51:22 admin interface at http://ubuntu2110:3000/admin
2022/07/06 09:53:59 configuration changed
2022/07/06 09:54:57 newcall: system=1 talkgroup=5490 file=20220705_135623Provincial_Public_Safety_Telecommunications_Network_005_(Regina_SaskPower_HeadOffice)_TRAFFIC__TO_5490_FROM_225.m4a success
2022/07/06 09:54:57 downstream: system=1 talkgroup=5490 file=20220705_135623Provincial_Public_Safety_Telecommunications_Network_005_(Regina_SaskPower_HeadOffice)_TRAFFIC__TO_5490_FROM_225.m4a to http://zion.hjl.ca:3000 success
```

So I'm really unable to reproduce your issue despite having tried almost everything.

Would it be possible for you to repeat the steps above with a new VM and test as I did?

I will flag this issue as not reproducible for now.

[20220705_135623Provincial_Public_Safety_Telecommunications_Network_005_(Regina_SaskPower_HeadOffice)_TRAFFIC__TO_5490_FROM_225.zip](https://github.com/chuot/rdio-scanner/files/9055476/20220705_135623Provincial_Public_Safety_Telecommunications_Network_005_.Regina_SaskPower_HeadOffice._TRAFFIC__TO_5490_FROM_225.zip)