chuot/rdio-scanner #197: Http Failure Response

dgwood12, 2022-05-29:
Hello,

I keep getting errors anytime I save my config via the admin interface. Anytime I make any kind of change and hit save, the rdio-scanner service in the background locks up for around 20-30 seconds and will fail to play any calls during that time until I am presented with an error that reads "Http failure response for http://ubuntu-sdr:3000/api/admin/config: 0 OK". Once the error displays, any call that came in during the lock up time starts to play. During this lock up time, you can still open the native app or web app and go to the different screens like talk groups, calls, etc. I have looked at the terminal output in these instances a couple of times and it spits out a bunch of dirwatch.walkdir errors saying that no such file or directory is found for some of my systems (this is correct as some of my systems do not have a directory yet as nothing has been recorded for those systems yet). Is this issue related to that? If I had directories for all my systems would it eliminate these errors?

A separate possibly related but probably unrelated issue I am having is with the Delete After option. I have this enabled in the dir watch options but I've never actually seen rdio-scanner delete any files that trunk-recorder is recording. If I am understanding the option correctly, rdio-scanner should be deleting the files that trunk-recorder generates after they are ingested into rdio-scanner, right?

Here's some additional info on my setup:
Using trunk-recorder and rdio-scanner on ubuntu 21.04. On the latest version of trunk-recorder and using the latest version of rdio-scanner (6.3.6).

dgwood12, 2022-05-30:
I have quite a few talkgroups (several hundred) but I am not sure that's the issue. I started my setup on a raspberry pi with the same number of talkgroups and that instance of rdio-scanner was working fine. It wasn't until I moved my setup to my home server on a linux VM runnin on unraid that the problem started. I'll send you a copy of my config json rght now.

dgwood12, 2022-05-30:
Ok. Sent to the email listed on your github profile.

dgwood12, 2022-06-01:
I tried shift-reload of the admin page as well as updating to 6.4.0 and the issue remains. I guess I can live with it as things still work and I have my system pretty much all the way set up now. Its just an annoyance when saving the config at this point because it crashes for about 30 seconds and then starts up again.

chuot, 2022-06-01:
Does it crash or did it freeze?

If it crashes, try to get the crash logs with `journalctl -u rdio-scanner.service`.

If it's frozen, try `kill -3` on the process id. This will generate a crash log which can be helpful in identifying the root cause.

dgwood12, 2022-06-05:
It freezes... Sorry, that's what I meant in my precious comment. I'll give that a shot. Thanks

chuot, 2022-06-13:
@dgwood12 Have you had a chance to test with the lastest version?

dgwood12, 2022-06-23:
I just tried on 6.4.5 and unfortunately its still doing it. It must be something with my config...

Its just a minor annoyance at this point since my system is all set up i'm not changing things very often anymore.

chuot, 2022-06-29:
I tried to reproduced your issue but did not succeed to.

Please try with the v6.5.1 which is just out. If you still have the issue, please post all the logs with the error messages. Unfortunately, I can't help much without them.

dgwood12, 2022-06-30:
I still got the error on v6.5.1. How can I get the logs to you?

chuot, 2022-07-04:
Simplest way is to start rdio-scanner from the command line, let it run until it freezes or when it shows the HTTP 0 OK error on the administrative dashboard (which shouldn't anymore), then `kill -3` the PID of rdio-scanner, grab all the stack trace and post it here (as an attachment).

chuot, 2022-08-19:
Closing for now as I have no stack trace to work on.

closed 2022-08-19 by chuot