chuot/rdio-scanner #198: rdio-scanner iOS not playing clips in "Search Call"

mlewis47, 2022-05-30:
Unable to play past calls on the iOS app.

"Live Feed" works.

"Search Call" brings up call list but clicking play just produces a spinning arrow.

Rdio Scanner native 1.6.0

chuot, 2022-05-31:
The fix has been applied to the Rdio Scanner v6.4.0. Try updating your server.

melbournefan, 2022-06-12:
I'm having the same issue on the PWA on V6.4.2, where the arrow would spin really fast and skip to the next 'suitable' recording despite the affected files been intact. There isn't really anyway either to determine which file gets skipped either

chuot, 2022-06-13:
The progressive web app on IOS does not work as intented, hence why the native app exists. Note that if you are not converting audio files to m4a, which is done by Rdio Scanner if configured so and if ffmpeg is available locally to Rdio Scanner, there's a big chance that they won't be played by the PWA.

When this occurs, look into the javascript console for any error messages. There should be some if the WebKitAudioContext cannot played some audio files because they can't be decoded.

@mlewis47 can you confirm that your issue as been solved with v6.4.0+ and close the issue occordingly.

closed 2022-07-11 by chuot