chuot/rdio-scanner #199: Database Inconsistencies

kcwebby, 2022-05-31:
We have over 100 instances; we are unable to determine why, but some of our databases (mysql) have the rdioScanners.Accesses field as a JSON fieldtype, and some of them as TEXT;  While direct manipulation of the database is not recommended; it was recently required due to an unrelated system error on our part, and the restoration of users/passwords from command line taking approximately 45seconds to 5 minutes per user, per rdio instance.

kcwebby, 2022-05-31:
Will do on the -cmd
The field type remains the same between all versions of 6.3.6 (we are all updated now to that level)
We will try 6.4.0 but it might make sense to do a database migration step to update / further check that the schema is all the same.  I can give exports if needed.

Is there a spot in the database that stores the server version, or schema version for review?  I can query that pretty quickly to see if a migration step was missed on some (because we jumped over versions on our way up to 6.3 from 5?

chuot, 2022-06-01:
The `rdioScannerMeta` contains the name of each schema which you can reference with waht's in the `database.go`.

closed 2022-07-11 by chuot