chuot/rdio-scanner #205: Feature Request: 12 Hour Clock Option

dgwood12, 2022-06-05:
For American users like myself, it would be nice to have an option to toggle between a 24 hour and a 12 hour clock with AM/PM showing for all places that currently show a time (call log, current time, last call time, etc

chuot, 2022-06-06:
I will add an option for this in a future version.

It is in 24H format mainly because it saves space on the screens. Having an `am` or `pm` in the history list and on the search screen would be repetitive and take up space for other data like the talkgroup name. This may not be a problem on a browser but on a smartphone.

dgwood12, 2022-06-07:
Thank you. Even if its just an A or P would be helpful.

Bobabadabouskie, 2022-06-08:
Just because you are American means you can't use 24hr format?

dgwood12, 2022-06-08:
I can but in America we mostly use 12 hour format. Like I said...would be nice to have...didnt say it's absolutely critical to add this...

chuot, 2022-06-13:
Closing this issue as the feature as been implemented

closed 2022-06-13 by chuot

dgwood12, 2022-06-30:
Thanks for implementing this. I am still seeing the 24 hour time for the current call time (under current call tag) 
as well as the time for the previous 5 calls on the android native app's main screen. The current time in the upper left corner is displayed correctly in 12 hour format as well as in the times main call log (screen you get to when you hit search call). I am guessing this is an update that is needed for the native app as the web interface is displaying the time correctly in all places. Here's a screenshot:

[image]

dgwood12, 2022-07-02:
Somehow today this mysteriously fixed itself. I am not sure if its something you did in the backend but it's now showing the 12 hour time for the current call and in the call log. One thing I did notice though is in the call log, all calls that take place during the 12 AM or 12 PM hour all show up as 12:00A or 12:00P despite them not actually taking place at the top of the hour. In the screenshot below, the first call in the log that shows as 12:00P actually took place at 12:06PM and the last call in the log that shows 12:00P actually took place at 12:37PM. Once calls get outside the 12AM or 12PM hour, the time is correct.
[image]

chuot, 2022-07-04:
The webapp wasn't updated yet when you installed the version that had the 12H format feature. Shift-reload forced the webapp update where it then displays the 12H format.

As for the 12H format issue on the native app, it will be fixed for native v1.8. It's a bug indeed.