chuot/rdio-scanner #206: SDRTrunk Metadata Dirwatch Parser Broken in 6.4.1

jslitzkerttcu, 2022-06-08:
It looks like 6.4.1 adds some code to the parsers.go (lines 130-134) to read the Album metadata information.  SDRTrunk applies the channel name to the album metadata and in trunked P25 systems, the channel name is called "TRAFFIC", so all my Talk groups are now ingesting as "TRAFFIC".  

Even though I have a dirwatch mask configured as "#DATE_#TIMEOKWIN_Tulsa_TRAFFIC__TO_#TG_FROM_#UNIT" and my example file name is "20220608_181135OKWIN_Tulsa_TRAFFIC__TO_1921_FROM_2172.mp3".

The metadata and file naming are not configurable (that I'm aware of), so I have always used the mask to accommodate the naming, but it looks like the code change may have impacted this.
[image]
[image]

See screenshots attached.

wmg819, 2022-06-09:
Having same issue it seems no matter what is in the talkgroup (blank or otherwise) name space it is being replaced with TRAFFIC in the meta data and database. The dirwatch is set to sdrtrunk with the defaults set for the mask.

jslitzkerttcu, 2022-06-10:
> Is the `name` field empty in the playlist editor for your channel (channels tab)?
> 
> I did indeed added something in 6.4.1 so the talkgroup name wouldn't be equal to the talkgroup label every time. It looks like when the `name` field is empty it's being replaced by `TRAFFIC` in the meta data.
> 
> Note that the dirwatch mask is not used at all when dirwatch type is set to sdrtrunk.

Sorry, I should have clarified - my second screenshot was the mp3 produced by SDRTrunk before Rdio-Scanner parsed it.  I'm updating to 6.4.2 now and will advise the results!

jslitzkerttcu, 2022-06-10:
Thank you for the prompt update on this, Chrystian!  That resolved the issue for me - I'm closing this out as resolved.

closed 2022-06-10 by jslitzkerttcu