chuot/rdio-scanner #215: Groups Missing from Select TG Screen

tadscottsmith, 2022-06-14:
Something happened to my DB, and no matter what I do, I can no longer get the talk groups to display by groups in the "Select TG" screen. This seemed to happen out of the blue on V6.2.4, so I updated to current and version, wiped everything, and tried to reimport the config as well as reimport from CSV. I can not get the "Groups" to show in select TG even though they appear to be recognized by the DB.

[image]
[image]
[image]
.

chuot, 2022-06-14:
Groups will only appear if they are linked to a talkgroup that the access code gives access to it.

Do you use access codes?

tadscottsmith, 2022-06-14:
I apologize, I thought that the Select TG had the talk groups separated and sorted by the group name, but it seems that hasn’t been the case. I will close.

closed 2022-06-15 by tadscottsmith