chuot/rdio-scanner #216: Unable to download files via Andorid (Premium version)

LoveScanner, 2022-06-14:
I can not download recordings through a premium version,
It is important to note that I have given permission for file permissions
 
I press the download button and nothing happens - maybe it's going down in the background? In which directory should I look for it?

chuot, 2022-06-14:
The first time you try to download, you should have been prompted to allow the storage permission to the app. It should ask again if you've answered deny. But on some android devices, you can also choose deny and do not ask again. If it's the case, you won't be prompted anymore and that could lead to the symptom you're describing. Go see in the app persmissions if it as storage permission or not; it should.

On android, calls are saved in the Download folder of the internal storage.

I have added a new error message to the upcoming version of the native app which will popup if the app doesn't have storage permission.

The audio issue where it stops after some time is know and is already logged in issue #191. The upcoming version will have a whole new audio subsystem which address that.

LoveScanner, 2022-06-14:
> The first time you try to download, you should have been prompted to allow the storage permission to the app. It should ask again if you've answered deny. But on some android devices, you can also [...]

Permission enabled, I also tried to reinstall and confirmed again did not solve the problem - for information

chuot, 2022-06-14:
Yes, please send the video with the details of your phone manufacturer, OS version and if the bootloader is still locked. My email is shown on my github profile.

LoveScanner, 2022-06-14:
> Yes, please send the video with the details of your phone manufacturer, OS version and if the bootloader is still locked. My email is shown on my github profile.

Done.

LoveScanner, 2022-06-14:
> Evereything looks good on your device.
> 
> Do you run the latest version of Rdio Scanner opensource (v6.4.4) ?

I had 6.4.2 now I updated to 6.4.4 - the issue was not resolved
If you want to diagnose a malfunction I have no problem sending you Anydesk remote access via mail

chuot, 2022-06-14:
It looks like /storage/emulated/0/Download on your Note 10+ with oneUI 4.1. I'm working on it.

chuot, 2022-06-20:
Ok, I found the issue where path.Base() doesn't work on Windows and should be replaced by filepath.Base().

It will be fixed in the next version.

closed 2022-07-11 by chuot