chuot/rdio-scanner #229: Date Display

Dewey3, 2022-07-01:
I know you want the screen to mimic an actual scanner as much as possible and that screen real estate is precious, but would it be possible to add the transmission date before or after the screen time?  This comes into play when reviewing recorded transmissions that span multiple days.  The date of the transmission being played back can be viewed by watching the search screen, but it would be nicer if that were also shown on the screen during playback.

chuot, 2022-07-04:
What about a `+1d 08:25` for calls older than 1 day, `+2d 08:25` for calls older than 2 days and so on?

Dewey3, 2022-07-06:
First, thank you for entertaining this.  A history of the request... I have Rdio set to delete transmissions older than 30 days.  There are times that I like to go in and play back old transmissions that were either unknown, or specialized/very little activity.  For this reason, there may only be 10 or 20 transmissions for the entire 30-day period (sometimes more during special events), so when playing back, I can not always tell what day the transmission occurred unless I go back into the search page and see which transmission is playing back.  With that said, I would probably continue to go back into the search page to see what is playing rather than doing the +/- math.  Either way, I again thank you for entertaining the request.

chuot, 2022-07-07:
Good point about doing the math. Then the same format as on the search page can be displayed only if the call being played is from the day before or older. I think I can fit that in the upcoming v6.5.2.

chuot, 2022-07-11:
Added in v6.5.2 and native v1.8.0

closed 2022-07-11 by chuot