chuot/rdio-scanner #231: Feature Request: Display of the site number on the application screen

LoveScanner, 2022-07-02:
It will greatly help me and others if it will be possible to see the site number while broadcasting the recording on the display screen

The data appears in the file name
[code]
2022/07/02 23:43:29 newcall: system=1 talkgroup=77 file=20220702_234327SystemName_**1.99**_TRAFFIC__TO_77_FROM_288884.mp3 success
[/code]

chuot, 2022-07-04:
You can put whatever you want in the system label field, just add the system id to it. In your case, because you're using SDRTrunk, change the system label on Rdio Scanner and on SDRTrunk / Playlist Editor / Channels / Name.

For example, say you system name is 'SERAM', just change it to 'SERAM 8086' where the last part is the system id.

chuot, 2022-07-07:
There is no site_id property for calls on v6 and it might be added in v7, but it's not sure yet.

Until then do what I do on my setup where I have split each site into its own system like RSP25MTL1 for site 1, RSP25MTL2 for site 2 and so on. Then I record each talkgroup from their main site *only* so that there are no duplicate calls (with sometimes different timestamps).

closed 2022-07-11 by chuot

LoveScanner, 2022-10-11:
The problem I have on the same system is 10+ sites