chuot/rdio-scanner #235: dirwatch not reading correctly

kb2ear, 2022-07-11:
I have a dirwatch that has filenames formatted like 20220710212553.WAV I have the extension set to WAV and the mask set to #DATE#TIME  But all recording are being imported with a 12/31 date and 19:03 time. They are being deleted every hour by the pruning.

This setup works in 5.2.9. but no 6.5.1

```
Jul 10 21:37:53 repeater Rdio Scanner master[17288]: 2022/07/10 21:37:53 newcall: system=42702 talkgroup=27089 file=20220710213727.m4a success
```

chuot, 2022-07-11:
I'm surprised it was working with v5.2.9 because it is the same exact regexp I'm using in the lataest v6. Anyway, I tried and indeed it isn't working. I will change the regexp to take your filename format into account, without breaking things for others.

chuot, 2022-07-11:
@kb2ear please test with v6.5.2, which is just out, and close the issue accordingly.

kb2ear, 2022-07-11:
looks like it's working