chuot/rdio-scanner #237: Feature Request: Info Section

dgwood12, 2022-07-13:
It would be great to have a button on the main screen that says info or something where the host of the scanner could post information like details about their scanner setup, current systems that are being monitored or even common radio codes and their definitions so users can easily look them up. Keep up the great work!

cpg178, 2022-07-20:
I would love to see something like this implemented too,

canuckcam, 2022-08-02:
+1 - could be simply a "help" button on the lower left perhaps, that can be redirected to a URL as a popup/new window or something? Policescanner.ca has (or had?) the rdio-scanner interface inside a frame so the surrounding was info and such.

chuot, 2022-08-08:
Well, on second thought, I won't add such a feature just because I want to keep the native apps the same as open source web app.

The problem with the native apps is that I can't provide a button that will open a random page with random items as they might break the agreement I signed with Google and Apple. They have more restrictions and checks when an app that can open web pages is submitted to their respective store, and I want to avoid that.

However, you can achieve the same result using a webpage with an `<iframe>` to Rdio Scanner. Keep an eye for the next release, I'll add an example for this in the `docs/examples/iframe` folder. You can then create your own page from this example.

closed 2022-08-08 by chuot