chuot/rdio-scanner #240: Duplicate Call Detection by Units

LoveScanner, 2022-07-17:
System that detects duplicate calls,
At the moment it only works by matching Talkgroup,
have importance the also matching by radio ID (Unit)
 
Thanks
good week!

chuot, 2022-07-18:
Not before version 7 because a database change is required. Will look at it then. Thanks

LoveScanner, 2022-08-18:
I see this problem too many times.

I ask myself, why required a change in the db, the data is already in the DB
I think the solution is relatively simple just to add in the query the "CheckDuplicate" function (source = source) in call.go
I would try it myself I just don't know how to compile code myself

chuot, 2022-08-19:
Is your `Duplicate Call Detection Time Frame` option set the the default value, which 500ms?

LoveScanner, 2022-08-19:
> I see this problem too many times.
> 
> example: `2022/08/18 10:06:47 newcall: system=1 talkgroup=7007 file=20220818_100644Server01_TRAFFIC__TO_7007_FROM_2212144.mp3 success`
> 
> `2022/08/18 10:06:48 newcall: system=1 talkgroup=7007 file=20220818_100645Server01_TRAFFIC__TO_7007_FROM_1140055.mp3 duplicate call rejected`
> 
> (I understand that pressing PTT for a second without sound, but still sometimes it can hurt long recordings)
> 
> I ask myself, why required a change in the db, the data is already in the DB I think the solution is relatively simple just to add in the query the "CheckDuplicate" function (source = source) in call.go I would try it myself I just don't know how to compile code myself

2200.

chuot, 2022-08-22:
Try to put back the default value of 500 and see if it helps.

LoveScanner, 2022-08-23:
> Try to put back the default value of 500 and see if it helps.

It doesn't solve my problem
I use rdio-scanner to receive simultaneous recordings from several computers (via a network)
And there are recordings that arrive 1-3 seconds late

The most perfect solution is that a test will also be carried out according to RadioID

chuot, 2022-08-23:
Don't record the same talkgroups from different sites as they sometimes don't have the same timestamps, which bugs Rdio Scanner's basic call duplicates detection. This is the case with the [RSP25MTL system](https://www.radioreference.com/db/sid/7537) that I record here. To mitigate the problem, I treat each site as a different system and record talkgroups from their primary sites only. Some talkgroups are recorded from site 1, which I ingest into system RSP25MTL1, other talkgroups from site 3 are ingested into system RSP25MTL3 and so on. I recommend you try the same technique with your setup. Checking source ids each time would reduce ingest performance drastically, as with trunk-recorder, for example, source ids are stored in JSON format, which is difficult to fit in an SQL query on SQLite. Also, checking at source ids when timestamps differ by up to 3 seconds would lead to other issues.

israelgoldstein, 2023-03-30:
I also like the suggestion.