chuot/rdio-scanner #241: System duplicated upon restarting server

cpg178, 2022-07-21:
I have noticed the last 2 times the machine running rdio-scanner reboots, or when I updated to 6.5.2 upon reboot a system named identical to one of mine, but with some extra talk groups is duplicated and takes up system id 2. The is that the systems that were previously ID 2 and 3 are bumped to 3 and 4 respectively, but new files being ingested sometimes do not recognize this change and get sent s to wrong place. This is easily fixed by deleting the errant system, but I'm confused as to why it keeps reappearing when I delete it and press save. 

This error has been appearing recently when changing the config, not sure if its related.
[image]

cpg178, 2022-07-21:
It appears after doing some research that even though I was removed the system from rdio-scanner, it was not removing it from the database.

chuot, 2022-08-19:
Fixed in v6.5.3

closed 2022-08-19 by chuot