chuot/rdio-scanner #255: Injested audio has tail end cut off

kcwebby, 2022-08-19:
When ingesting audio (dirwatch if it matters)  the tail end of the audio file is cut off, leaving some of the recorded audio behind.
[Audio Sample.zip](https://github.com/chuot/rdio-scanner/files/9384353/Audio.Sample.zip)

This is done with "audio conversion" set to disabled.   This happens with every single audio file.  

It does not seem to get further trimmed if passed to a downstream RDIO server.

chuot, 2022-08-19:
Are these files from ProScan?

kcwebby, 2022-08-19:
Not sure if its relevant, but this is Proscan / RDIO for windows.  6.5.3

chuot, 2022-08-19:
I have an issue with ProScan where it sometimes continues to write the file after a long pause. I need to find a way around this to know when the file is really full and done.

kcwebby, 2022-08-19:
So would a good work around be to expand the "pick up" delay from 5000 ms to 7500 or 10000 ms?

chuot, 2022-08-19:
Some users set it up to 60 seconds, but that defeats the purpose of the dirwatch. Again, issue #254 and issue #255 seems to be related.

chuot, 2022-09-01:
Can you try to configure ProScan to create `wav` files instead of `mp3` file and see if it helps? You will have to change the file extention in your dirwatch definition accordingly.

cpg178, 2022-11-01:
This is related to an issue I had where ProScan constantly appends the recorded file during the recording, unlike SDRTrunk which does not write the file to the directory until the end of the recording. I was able to work around this issue by enabled the proscan option to append the end time of the recording which then adds 'TO (endtime)' to each file once it is COMPLETE. I then have a script which scans a folder that ProScan sends too and only moves files containing 'TO' (Complete recordings) to another directory that Rdio has a dirwatch on. It would be cool to add the ability for the Rdio dirwatch to only pull in file names that contain a certain string such *TO* as that would not only fix this issue that kcwebby and other proscan users have, but also remove the need for the complicated workaround.

kcwebby, 2022-11-02:
Interesting concept..  would you mind sharing the script you are using as a
workaround?

closed 2023-03-12 by chuot