chuot/rdio-scanner #257: Feature Request: Command line param to export the configuration to a JSON file

atlex, 2022-08-23:
Hi @chuot,

We all need backups :) Automatic backups.
It would be nice and safe to have automatic backup of the rdio-scanner configuration.
Something like that:

$ crontab -e

```
0 2 * * *  /home/pi/rdio-scanner/rdio-scanner --export rdio-scanner_20220823.json
```

chuot, 2022-08-23:
You can already. Create a script like this:

```bash
#!/bin/sh
cd /home/rdio-scanner
RDIO_ADMIN_PASSWORD=yourpassword ./rdio-scanner -cmd login
./rdio-scanner -cmd config-get +out backup.json
./rdio-scanner -cmd logout
```

and crontab it.

`./rdio-scanner -cmd help` will give you more details:

```bash
$ ./rdio-scanner -cmd help

Available Commands:

  admin-password – Change administrator password.

                $ ./rdio-scanner -cmd admin-password +password <password>

  config-get  – Retrieve server's configuration.

                $ ./rdio-scanner -cmd config-get +out <file.json>

  config-set  – Set server's configuration.

                $ ./rdio-scanner -cmd config-set +in <file.json>

  login       – Login to server.

                $ RDIO_ADMIN_PASSWORD=<password> ./rdio-scanner -cmd login
                $ ./rdio-scanner -cmd login +password <password>

  logout      – Logout from server.

                $ ./rdio-scanner -cmd logout

  user-add    – Add a user access.

                $ ./rdio-scanner -cmd user-add +ident <ident> +code <code>

                Optional:

                  +expiration <RFC3339 format>      – User access expiration date.
                  +limit      <limit>               – Concurrent user access limit.
                  +systems    <sysid1[,sysid2,...]> – Specific system access.

  user-remove – Remove a user access.

                $ ./rdio-scanner -cmd user-remove +ident <ident>

Global Options:

  +token      – Session token keystore. Default is `.rdio-scanner.token`.
  +url        – Server remote address. Default is `http://localhost:3000/`.
```

atlex, 2022-08-25:
My full script

    $ nano rdio-scanner-backup.sh
    
```
#!/bin/sh

RDIO_HOME=/home/pi/rdio-scanner
BACKUP_HOME=$RDIO_HOME/backup
BACKUP_DATE=$(date '+%Y%m%d')

cd $RDIO_HOME
RDIO_ADMIN_PASSWORD=MYPWD ./rdio-scanner -cmd login
./rdio-scanner -cmd config-get +out $BACKUP_HOME/rdio-scanner_$BACKUP_DATE.json
./rdio-scanner -cmd logout
```
    $ mkdir backup
    $ chmod +x rdio-scanner-backup.sh
    $ ./rdio-scanner-backup.sh

    $ crontab -e

```
0 2 * * * /home/pi/rdio-scanner/rdio-scanner-backup.sh
```

chuot, 2022-09-01:
@atlex Thanks for sharing your script. I guess you can now close this issue.

closed 2022-09-09 by chuot