chuot/rdio-scanner #259: Talkgroups cut off after restart

dgwood12, 2022-09-01:
I recently added a new system that has many talk groups (326 to be exact) and am encountering a weird bug when I restart rdio-scanner. I imported the talkgroups via the import tool and everything imports great. Once imported I finished the system set up and got everything working beautifully...all the TGs were showing in the select TG section in rdio scanner and everything comes through loud and clear. The issue I am having is whenever rdio-scanner restarts, around 128 of the talkgroups for that system disappear leaving me with only 198 of the 326 I imported. This only happens when rdio-scanner is restarted. 

I know its not a total talkgroup limitation as I used to have this one system split up into 11 different systems. I had to recombine them all into one system since trunk-recorder would crash every time the control channel changed due to it trying to cycle through the control channels for 11 systems at one time.

I have recreated the system several times now and every time rdio-scanner restarts, the talk groups get cut off at the same place.

I have attached my talkgroups file for reference.

I am using v6.5.3 on Ubuntu 21.04 (AMD64).
[sacramento-county-srrcs-talkgroups.csv](https://github.com/chuot/rdio-scanner/files/9472394/sacramento-county-srrcs-talkgroups.csv)

chuot, 2022-09-02:
Check your logs just after importing your talkgroups, there must be something odd that triggers an error when saving the data to the database.

dgwood12, 2022-09-04:
> Check your logs just after importing your talkgroups, there must be something odd that triggers an error when saving the data to the database.

I'm out of town but I'll check in a few days and report back

dgwood12, 2022-09-08:
I get the following error when saving the config after creating a new system with the talkgroup import:

2022/09/08 15:10:05 admin.confighandler.put: systems.write: constraint failed: UNIQUE constraint failed: rdioScannerSystems.id (2067)

chuot, 2022-09-09:
Try this: remove the system from your configuration and make sure to save the changes. Then reload the entire CSV file with all talkgroups. Be sure to set back the system ID before saving the new configuration. Alternatively, you can simply rely on the AutoPopulate feature which will automatically create missing talkgroups on call ingestion. This is the best way to do it IMHO.

If it still doesn't work after trying the steps above, email me your exported configuration, I'll try to reproduce the issue on my dev box and see what I do.

dgwood12, 2022-09-09:
Thank you! I got it working now. I think the issue might have been that I was using a system ID that wasn't currently being used but was used in the past? Not sure but I decided to use an ID that I know has never been used for any other system (99) and it worked. All talkgroups remain after save and restart. Thanks again!

closed 2022-09-12 by chuot