chuot/rdio-scanner #266: Rejected calls due to system not existing

cpg178, 2022-09-26:
I have been having an intermittent issue where calls from SDRTrunk for a particular system would be rejected as it would try to send them to sysid 4 which did not exist. Normally they would route to sysid5. This got annoying so I decided to fully wipe my DB and start from scratch by setting up my 4 systems with the system name the exact same as SDRTrunk but now it is trying to send the calls to sysid 5 which does exist... any insight will be helpful as if I turn on the main auto populate option it makes a new system for each call that comes in, even if the TG is the same.
[image]

cpg178, 2022-09-26:
So some further info, I changed the id of the system to 5 to see if that would fix it, and its now trying to send all of the files to id4 (it was 4 originally, before I switched it to 5).

cpg178, 2022-09-26:
User error, auto populate was not checked WITHIN the system

cpg178, 2022-09-26:
So apparently enabling auto populate, does not fix the issue either. I tried changing the system ID to 5 as that was what rdio was trying to send the files  to, but then the system starts trying to send it 4 which rejects it as 4 no longer exists.

chuot, 2022-09-27:
Have you checked discussion #262 that you opened on the same topic. I gave you some hints in this regard.

cpg178, 2022-09-27:
Yes, both are named ‘Luzerne’. I turned on auto populate to allow it create
a new system yesterday afternoon, and seemed to stop the error from
occurring on the new DB that I had created. I will continue to monitor to
see if the intermittent errors occur like they did before, but am hopeful a
DB wipe will fix it as my old DB the ids were not sequential for
some reason.

chuot, 2022-09-27:
The SDRTrunk parser relies strictly on the MP3 tags generated by SDRTrunk. Unfortunately, there is no system id available there, only the system name that matches what is set at the Playlist Editor level.

So the parser code [here](https://github.com/chuot/rdio-scanner/blob/af41604322ba3be3e0843263472a0413a8badff7/server/parsers.go#L188-L193) searches for a matching system name. If it finds it, it uses the system ID to populate the call data. If it does not find it, it tries to find the next available system ID, which will then be refused if autopopulate is not enabled.

cpg178, 2022-09-27:
Thanks for the insight with the code, it’s interesting as I have 2 other
separate SDRTrunk systems that had no issues just this one. The other 2 are
NBFM so not sure if SDRTrunk would name the audio files different from the
trunked system. What was especially interesting was it would reject TGs
that already had been ingested to rdio before, saying no system/tg existed,
prior to me wiping the DB and starting over.

cpg178, 2022-09-27:
> Maybe something is wrong with the MP3 tags? Try to identify a file
> outputed by SDRTtrunk and check the MP3 tags for some hints. Beware that
> Rdio Scanner overwrite those MP3 tags if you turned on the audio conversion
> and if ffmpeg is installed along with Rdio Scanner. In that case, check the
> files from SDRTrunk directly, not from Rdio Scanner.

WIll do, I tried to look for any settings to customize the file names like
with ProScan yesterday but could not find any.

cpg178, 2022-09-27:
Here is a metadata extraction from a recording along with a screencap of my system info, coincidently it has just started to reject certain files now after working flawlessly for almost 24hrs. 
[image]
[image]

chuot, 2022-09-27:
Great, that will help in troubleshooting your issue. If possible, can you send me by email a Rdio Scanner config export of your instance.

cpg178, 2022-09-27:
Just sent it to you

chuot, 2022-10-04:
@cpg178 Please update the issue with your findings in regards to multi-site recording.