chuot/rdio-scanner #276: Run custom command if system has not ingested a call in X hours

rml596, 2022-11-20:
Similar to how Broadcastify sends emails that a node is offline, it would be nice if Rdio-Scanner could send a notification that a System has not received a call in X amount of time. This should be configurable and optional. 

It would be great if Rdio-Scanner could connect to a SMTP server, but I see the easiest way to achieve this would be running a script. That way the user could build a custom script to send an email, push notification, etc. 

One additional implementation would be to connect Trunk-Recorder to Rdio-Scanner via a web socket (possibly via Trunk-Recorder’s Status-Server) to check for Control Channel decode errors of a p25 system. This implementation is probably out-of-scope, but is worth mentioning.

chuot, 2022-11-21:
I'll see want in can implement in v7, which is due early 2023. Some kind of a watchdog feature could be nice indeeed.

Thanks for submiting the idea.

tadscottsmith, 2022-11-22:
Great idea!

Clark985, 2023-04-05:
If this could also fire off commands, that would be awesome.   SDRTrunk seems to glitch after a while (known memory leak issue), this could be another method to self-heal.