chuot/rdio-scanner #280: DirWatch New Trunk Recorder Missing System Field

adorsett, 2022-11-24:
Upgraded to 6.6.2 and built a new config as I am converting from call-upload to DirWatch. Tried adding the DirWatch in admin and it continued to show red !. Didn’t see an option to select System. Changed to “Default”, selected System, changed back to Trunk Recorder, red ! cleared and I could save config.

chuot, 2022-11-24:
Just to make sure I understand, you're using a trunk recorder with a rdio scanner. You were initially using an upload script to feed rdio scanner, but now you want to convert to dirwatch. This will only work with dirwatch type trunk-recorder as it will ingest metadata from the json file that comes with the audio file. The filename itself does not contain enough information about the system id. It may work with dirwatch type default, but yes, you have to specify the system id manually because the #SYS metatag won't work for the dirwatch.mask, because there is no such information in the filename, and you would also be stuck with a bunch of json files because the audio file itself would be deleted after ingestion, not the json file with type default.

If you could post screenshots of the definition of dirwatch which gives a red exclamation mark. This will help me understand where the error is.

Personally, I would use the rdio scanner plugin that comes with trunk recorder. It works great and it will be easier to migrate rdio-scanner v7 when it comes out. Unless you have a specific reason not to use it.

adorsett, 2022-11-25:
so maybe the bug is it’s looking for a System when it shouldn’t be?

Here’s the screen with the red ! That I’ve filled out but can’t save. 
[image]

I switched the type to Default which shows more options. 
[image]

I set the System
[image]

I changed the type back to Trunk Recorder and the red ! is gone and I can save. 
[image]

I took your advice and switched to the plug-in. I liked that DirWatch will import recordings that happened while Rdio Scanner is stopped. I’ve had problems with it where it stops working or whatnot and I lose recordings. They are all sitting there in the folders but the upload script only tries once. It’s after a period of time I notice there’s 29GB of audio clips that haven’t uploaded and I start digging into why. It would be nice if the plug-in would push any files in the archive every time TR starts or periodically retry.

chuot, 2022-11-25:
It was indeed a bug that I fixed with the 6.6.3 version which has just been released. Please test, confirm and close the issue. Thanks for reporting this issue.

closed 2022-12-09 by chuot