chuot/rdio-scanner #307: Unable to record LTR from ProScan

cpg178-kcd, 2023-01-03:
When trying to record LTR it appears RDIO does not like the way LTR formats the talkgroups with dashes. Is there another way around this, I am using the recommended proscan file formatting. 
[image]

chuot, 2023-01-04:
You will for sure have to adjust the ProScan formatting to match your dirwatch mask definition, or vice versa. What does your dirwatch mask look like?

cpg178-kcd, 2023-01-04:
My dirwatch mask is as follows #DATE_#TIME_#SYSLBL_#GROUP_#TGLBL_#TGMHZ with the ProScan file name set to be %YY-%M-%DD_%TT_%S_%G_%C_%FT which is Year_Month_Date_Time_System_Group_Channel_Talkgroup. However LTR has Talkgroups formatted like this '0-06-002' which I think is confusing rdio. Is RDIO smart enough to create its own TGs if I was to remove the TGID from the formatting?

cpg178-kcd, 2023-01-04:
I have analog channels on the same device that pull through no issue, although it appears proscan takes the frequency and makes that the TG. That would not work for LTR as since it is trunking, many talkgroups can be using the same frequency.

chuot, 2023-01-29:
@cpg178-kcd Any progress on this ?

cpg178-kcd, 2023-01-29:
There has not, Rdio was still not pulling it in even when set to TG AFS, I did do a temporary workaround by setting the home repeater frequency up in SDRTrunk but that means missed and unwanted calls as if the home repeater is in use it will kick the TG over to another frequency

cpg178-kcd, 2023-02-06:
So I gave this a shot today, see below for my config info, but I am now getting an error that the TG does not exist. I went to try to manually create the TGs in the Systems tab but it wont let me do it in AFS format 0-06-02. So in the meantime I have set the DirWatch to direct everything into a shared TG on Rdio but this is not ideal. 

RDIO Mask: #DATE_#TIME_#SYSLBL_#GROUP_#TGLBL_#TGAFS
System 5 set as an AFS system on Options tab.
ProScan file format 
[image]

chuot, 2023-02-07:
In its current state, the latest version of Rdio Scanner does not interpret `0-06-002` as AFS when using the #TGAFS metatag. And the AFS system option is only for displaying the talkgroup ID in AFS format on the main screen.

I will modify the #TGAFS metatag in the upcoming v7 so that it can decipher talkgroup ids like "0-06-002".