chuot/rdio-scanner #312: Feature Request: Show UID # under "UID Name" field on main ui

infecticide, 2023-01-11:
It would be nice to see the UID number somewhere on the main window the same way we can see the TGID name and number.

infecticide, 2023-01-29:
The reason I requested this is for example:

- Fire department sometimes moves their radios around within a hall/between people or a truck unit is under repair.  In this case I already have a UID Alias setup but I can't see the UID in rdio-scanner to update it with the correct name.

Clark985, 2023-04-05:
Great suggestion, I've wished for the same feature.  In my case, my county is in the midst of deploying new radios and repurposing others.

McHuskey, 2023-05-01:
I imagine they are talking about the UID in the history section like this:

[image]

Clark985, 2023-05-01:
Normal behavior is to show the UID numerically unless it's defined as a unit.
When a unit is defined, the label is shown instead of the numerical UID.  So if you realize you have one labeled incorrectly, it's hard to determine which ID it was if they share the same name, for example.

UI real estate is probably at a premium, but the option to toggle the numerical UID as an additional field (in the presence of a unit label) would be welcomed.

doftheworld, 2023-05-01:
> I imagine they are talking about the UID in the history section like this:
> 
> [image]

I made a seperate post making this request, but yes, what you show there is exactly what I wish it did. I'm a little confused because I do not see that field now and it seems you do? I don't see any option to togle this on/off.

McHuskey, 2023-05-03:
That is just a mocked up image... it's not a feature that actually exists in rdio-scanner yet.

doftheworld, 2023-05-03:
> That is just a mocked up image... it's not a feature that actually exists in rdio-scanner yet.

Ahh! Well it had me fooled LOL! It's exactly what I am wishing for.